openhwfoundation/cv32e40p #458: custom-0 and ccustom-1 encodings legal/illegal use

darbaria, 2020-08-26:
According to RISC-V ISA spec v2.2., page 103: Major opcodes marked as custom-0andcustom-1will be avoided by future standard extensions and are recommended for use by custom instruction-set extensions within the base 32-bit instruction format.  

I am not sure if the current core is meant to use this space for PULP extensions. It appears that the core will consider it illegal if custom 0 and 1 encodings are encountered during decode if PULP_XPULP = 0. We have an exhaustive proof that matches this interpretation. When PULP_XPULP=1, illegal instruction is not flagged. This could be expected or maybe not. 

Assertion checks the relationship between custom-0 and custom-1 encodings and decode illegal instruction signal 

It fails for:

parameter PULP_XPULP = 1,
parameter PULP_CLUSTER = 0,

Changing the setting to:

parameter PULP_XPULP = 0,
parameter PULP_CLUSTER = 0,

proves the property exhaustively in formal.

Issue found by: Axiomise formalISA proof kit, and Cadence Jaspergold and Mentor Graphics Questapropcheck.

VCD for failing assertion is attached.
This is connected to the discussion being made on: #452
[custom_01_encoding.vcd.gz](https://github.com/openhwgroup/cv32e40p/files/5130303/custom_01_encoding.vcd.gz)

Silabs-ArjanB, 2020-08-27:
Hi @darbaria When using PULP_XPULP = 1 the core is _supposed_ to use custom-* encodings, see https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/instruction_set_extensions.html and #452 . As reported in #452 we intende to exclusively use such encodings for PULP instructions. Could you please close this issue if you agree it has been answered adequately or let us know what info is missing to answer your question?

Silabs-ArjanB, 2020-09-02:
We are closing this issue as we consider the question answered and as the related issue is fully captured in #452 . Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.  Thank you for contributing to this core; raising issues will help to improve it.

closed 2020-09-02 by Silabs-ArjanB