openhwfoundation/cv32e40p #459: ISA custom-2 and custom-3 encodings show mixed behaviour with different settings of RTL

darbaria, 2020-08-26:
According to RISC-V ISA spec v2.2., page 103: The opcodes marked custom-2/rv128 and custom-3/rv128 are reserved for future use by RV128, but will otherwise be avoided for standard extensions and so can also be used for custom instruction-set extensions in RV32 and RV64.

Without PULP extensions, the assertion that checks the relationship between custom-2/3 encodings and decode illegal instruction signal proves in formal. i.e., if these encodings are encountered the illegal signal is asserted.

Need to know what is the encoding space that is considered illegal when PULP extensions are enabled? When they are disabled, the behaviour of the core matches the expected ISA requirements.

It fails (possibly correctly) for:
[custom_23_encoding.vcd.gz](https://github.com/openhwgroup/cv32e40p/files/5130361/custom_23_encoding.vcd.gz)

parameter PULP_XPULP = 1,
parameter PULP_CLUSTER = 0,

Changing the setting to:

parameter PULP_XPULP = 0,
parameter PULP_CLUSTER = 0,
proves the property exhaustively in formal.

Issue found by: Axiomise formalISA proof kit, and Cadence Jaspergold and Mentor Graphics Questapropcheck.
VCD for failing assertion is attached.

This is connected to the discussion being made on #452

      
    
  

          

      

        
          

    

                

  
  

  
  

  

  
  

  
        
  
    

    

        @darbaria
  darbaria
  

      
changed the title
custom-0 and ccustom-1 encodings legal/illegal use

custom-0 and custom-1 encodings legal/illegal use

    now

  

  
  
      

            

            
            
                    
  
  @darbaria

davideschiavone, 2020-08-26:
Thanks @darbaria,

you are right,

parameter PULP_XPULP = 1,
parameter PULP_CLUSTER = 0,

it's a configuration that is not allowed. I am wondering how to solve this, we should probably add an assertion (@Silabs-ArjanB ?)

Silabs-ArjanB, 2020-08-26:
> 
> 
> Thanks @darbaria,
> 
> you are right,
> 
> parameter PULP_XPULP = 1,
> parameter PULP_CLUSTER = 0,
> 
> it's a configuration that is not allowed. I am wondering how to solve this, we should probably add an assertion (@Silabs-ArjanB ?)

Hi @davideschiavone Why would that not be allowed? That is our default configuration.

davideschiavone, 2020-08-26:
Sorry my bad, I read XPULP=0 and CLUSTER=1 , my bad

Silabs-ArjanB, 2020-08-27:
Hi @darbaria When using PULP_XPULP = 1 the core is _supposed_ to use custom-* encodings, see https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/instruction_set_extensions.html and #452 . As reported in #452 we intende to exclusively use such encodings for PULP instructions. Could you please close this issue if you agree it has been answered adequately or let us know what info is missing to answer your question?

darbaria, 2020-08-27:
My question is: are you covering the entire space of available custom encodings? If not, then what is it? Is this defined somewhere in the cv32 spec? In case you're not using the entire available encoding then what is the expected behaviour of the core when it encounters an encoding you didn't code explicitly as custom? From my discussion yesterday, it is clear to me that in the absence of any PULP extensions, the decoder module raises an illegal flag when encountering any decode. 

If the question is still not clear, we can go on a call to clarify this.

Silabs-ArjanB, 2020-08-27:
Hi @darbaria The entire custom encoding space is not filled. I would expect that we (should) generate illegal instruction exceptions for the non-filled custom space. In https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/instruction_set_extensions.html you can find the currently used opcodes. In the spreadsheet attached to #452 you can find the newly proposed opcodes (but that spreadsheet has not been committed to yet).

Silabs-ArjanB, 2020-09-02:
We are closing this issue as we consider the question answered and as the related issue is fully captured in #452 . Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.  Thank you for contributing to this core; raising issues will help to improve it.

closed 2020-09-02 by Silabs-ArjanB