openhwfoundation/cv32e40p #460: interrupt CSR checks fail when interrupt occurs during div instruction **Component:RTL**

strichmo, 2020-08-26:
When enabling random interrupts with the ISS in the core-v-verif testbench CSR miscompares can occur on instruction retirement boundaries.

The testcase is an interrupt that is acknowledge during a div instruction.  See the first figure below.

[image]

In this case the interrupt is acknowledge 2 instruction retirements before the machine mode interrupt vector is entered.  Therefore the PC is miscompared against the ISS.

To address this I added a staging of the "deferint" signal which masks the irq assertion in the ISS to the second instruction.  This enables the PCs to compare properly between RTL and the ISS.

[image]

However with this case, the CSRs for machine mode interrupts (mepc, mstatus, mcause) will miscompare on the first instruction retirement because the RTL updates the CSRs immediately upon interrupt acknowledgement.

### Steps to Reproduce
To replicate the first condition (PC miscompare):

1. Use this branch and fork:
https://github.com/strichmo/core-v-verif/tree/strichmo/irq_random_fail_0

2. Run this in cv32/sim/uvmt_cv32
% make corev-dv gen_corev_rand_interrupt_test NUM_TESTS=1 SEED=1
% make custom CUSTOM_PROG=corev_rand_interrupt_test_0 USER_RUN_FLAGS="+gen_irq_noise +UVM_MAX_QUIT_COUNT=20,NO" SIMULATOR=xrun

To replicate with the proposed "deferint_prime" fixes:

1. Use this branch and fork:
https://github.com/strichmo/core-v-verif/tree/strichmo/irq_random_fail_fix_0

2. Run the same tests:
% make corev-dv gen_corev_rand_interrupt_test NUM_TESTS=1 SEED=1
% make custom CUSTOM_PROG=corev_rand_interrupt_test_0 USER_RUN_FLAGS="+gen_irq_noise +UVM_MAX_QUIT_COUNT=20,NO" SIMULATOR=xrun

Silabs-ArjanB, 2020-08-28:
Hi @strichmo Thanks for filing this issue. I agree there is an issue here. I don't think the RTL in itself has an issue, but for sure the 'tracer' retire signal does not lead to the report of a consistent state. I am now thinking if we should fix this in the tracer or if we change the actual RTL such that the current tracer works correctly.

Silabs-ArjanB, 2020-08-31:
Hi @strichmo This issue is even worse than suggested. If you would have a CSR write (to e.g. mepc, mcause, etc.) instead of the DIV, then the actual update caused by the IRQ would get lost because if happens too early. In that case your work-around would not work either, see below waveform:

[image]

And in fact, the above waveofrm shows a second RTL bug as well, for which I will file a separate ticket (the CSR instruction completes before the lw instruction if the lw sees a stall as in my example:

lw x5, 0(x4)     
csrw mepc, 0x1e

Silabs-ArjanB, 2020-08-31:
@davideschiavone In above example what do you think the RTL fix should look like? A) Defer the moment at which the irq_ack_i based write to MEPC happens B) Only perform regular CSR writes in the first cycle such an instruction is in the EX stage

Silabs-ArjanB, 2020-09-01:
Decided not to change anything related to the DIV+IRQ example, but fixed the similar CSRW + IRQ example, which is an actual RTL bug.

Silabs-ArjanB, 2020-09-24:
Hi @strichmo Can you please close this issue if you agree it has been fixed?

strichmo, 2020-09-27:
Closing issue.  Random interrupt tests are running clean.

closed 2020-09-27 by strichmo