openhwfoundation/cv32e40p #462: Hardware Loop ssertion fires

Silabs-ArjanB, 2020-08-31:
The following assertion in cv32e40p_hwloop_regs.sv does not pass Formal (this was already reported in the PR review, but not fixed; likely assumes need to be made explicit)

    assert property (
      @(posedge clk) (valid_i) |-> ($countones(hwlp_dec_cnt_i) <= 1) );

Silabs-ArjanB, 2020-11-20:
[hwloop_assertion.vcd.gz](https://github.com/openhwgroup/cv32e40p/files/5571998/hwloop_assertion.vcd.gz)

pascalgouedo, 2022-05-17:
Due to following Hardware Loop constraint from User Manual, both counters can never be decremented at the same time.
So hwlp_dec_cnt_i can not be at 2'b11 but only 2'b01 or 2'b10.

- The End address of the outermost HWLoop (1) must be at least 2 instructions further than the End address innermost HWLoop (0), i.e. HWLoop[1].endaddress >= HWLoop[0].endaddress + 8. An illegal exception is raised otherwise.

closed 2023-01-04 by pascalgouedo