openhwfoundation/cv32e40p #466: CSR instruction retires out of order

Silabs-ArjanB, 2020-08-31:
In the following code (with the lw getting some wait states on rvalid), the later instruction (csr) retires before the earlier instruction (lw)

lw x5, 0(x4)      
csrw mepc, 0x1e

This at the very least leads to difficulties when doing state comparisons against an ISS (if not down right state corruption, although the shown example does not show an actual functional issue).

[image]

Maybe this issue is (partially) solved by https://github.com/openhwgroup/cv32e40p/pull/181 (I did not study that pull request).

Silabs-ArjanB, 2020-08-31:
This is by design; it is not an RTL bug. Whether this is 'workable' for the verification team is up to them; if this retirement strategy leads to verification issues, then we can re-open this issue and consider whether we adapt the 'tracer' or the RTL. If the verification team only compares the 'destination register' against the ISS for each instruction (as opposed to the entire register file + CSR state), then the above effect should not be noticeable.

closed 2020-08-31 by Silabs-ArjanB

davideschiavone, 2020-09-01:
The CSR instructions can be "fixed" by forcing the pipe to flush always, not only on special addresses of the csr.
The LW+ADD is by design yes. It's a WONT FIX because there is no real WB stage here...