openhwfoundation/cv32e40p #470: Improve interrupt latency

Silabs-ArjanB, 2020-09-01:
Interrupt latency on CV32E40P seems longer than needed. Specifically the following transition seems to add an unneeded cycle: DECODE -> IRQ_FLUSH -> IRQ_TAKEN_ID.

On a next core consider if a cycle can be saved here.

davideschiavone, 2020-09-03:
IRQ_FLUSH is needed to kill an interrupt in case of a a single pulse. But we can probably merge these 2 states

davideschiavone, 2020-09-03:
I made a PR for it. It passes the interrupt test in the cv32 folder but we should try with more advanced test. Pinging 
@MikeOpenHWGroup  for this

MikeOpenHWGroup, 2020-09-18:
Hi @strichmo, can you have a look? Its possible that the improved latency in the RTL will impact the integration with the Reference Model.

Silabs-ArjanB, 2020-09-18:
PR https://github.com/openhwgroup/cv32e40p/pull/508 also fixes #470

Silabs-ArjanB, 2020-09-18:
(Issue has not yet been fixed; pull request https://github.com/openhwgroup/cv32e40p/pull/508 has been withdrawn)

closed 2020-09-29 by Silabs-ArjanB

eroom1966, 2024-07-06:
@davideschiavone @MikeOpenHWGroup 
I know this issue was closed, but I had a question that is closely related to the original comment.

For the CV32E40P (v2) what is the worst case latency for an interrupt to be taken.
Consider the following

Interrupts are enabled in both the MSTATUS and MIE registers.
A local Interrupt becomes active on the interrupt line.

What is the worst case scenario described in either clock cycles or number of subsequent instruction retirements, before the processor updates its internal state and jumps to the address indicated by MTVEC ?

I cannot find anything in any documents under openhwgroup, and the closest information I have found is here
https://past.date-conference.com/proceedings-archive/2021/pdf/1782.pdf

I am sure there is some defined upper limit which would be critical to know for some real time applications, but trying to find that is difficult.

pascalgouedo, 2024-07-08:
Hi @LeeHoff 
It is always hard to characterize interrupt latency on a core.
You have to take hypothesis like fetch and load/store latency, succession of instructions...
And it will not be a maximum value but the best value you could have in normal situation.
Because for maximum value it could be few tens of cycles depending if you allow or not nested interrupts, those OBI latencies, ... .

So if OBI latencies are at the minimum (no wait-state), there are no multi-cycle integer instruction (like MULH/DIV) and MSTATUS.MIE and MIE are all enabled, an interrupt should be taken in few cycles (maybe 6 as mentioned in the article but I have to look at it to be sure).

Then the article makes the difference between the first instruction executed once the interrupt is taken and the interrupt routine first instruction really being part of the application. They even degrades it by 6 cycles considering you always want to have nested interrupts.

So finally maybe this 33 cycles is correct but this is a number which has some hypothesis behind.
You can't really use it during verification campaigns because all those hypothesis will not be followed for sure...

MikeOpenHWGroup, 2024-07-16:
Hi @pascalgouedo, I think you meant to address @eroom1966.