openhwfoundation/cv32e40p #471: Not able to launch hello world example

abettati, 2020-09-02:
Hi to you all, 
I'm trying to launch the examples in `cv32e40p/example_tb/core/`.
Here's what I did:

```shell
# clone core repo
git clone https://github.com/openhwgroup/cv32e40p.git

# download fpnew (necessary for proper build)
make deps
```

Modified make file in `cv32e40p/example_tb/core/` to match my system settings:

```make
MAKE            = make

# vsim configuration
#VVERSION                = "10.7b"
VVERSION                = ""

#VLIB           = vlib-$(VVERSION)
VLIB            = vlib$(VVERSION)
VWORK           = work

#VLOG           = vlog-$(VVERSION)
VLOG            = vlog$(VVERSION)

VLOG_FLAGS      = -pedanticerrors -suppress 2577 -suppress 2583 -define CV32E40P_TRACE_EXECUTION -define CV32E40P_ASSERT_ON
VLOG_LOG        = vloggy

#VOPT           = vopt-$(VVERSION)
VOPT            = vopt$(VVERSION)
VOPT_FLAGS      = -debugdb -fsmdebug -pedanticerrors +acc #=mnprft

# VSIM          = vsim-$(VVERSION)
VSIM            = vsim$(VVERSION)

#VSIM_HOME      = /usr/pack/modelsim-$(VVERSION)-kgf/questasim
VSIM_HOME       = /opt/questasim
```

Added toolchain to path: `export RISCV=/opt/riscv`

I try to launch the custom (hello world) example: `make custom-vsim-run` and get the following error
```
# Compiling /tmp/andrea.bettati@egguser-virtual-machine_dpi_25524/linux_gcc-5.3.0/exportwrapper.c
# Fatal: Fatal: (vsim-3828) Could not link 'vsim_auto_compile.so': cmd = '/opt/questasim/gcc-5.3.0-linux/bin/gcc -shared -fPIC -m32 -B/usr/lib32 -o "/tmp/andrea.bettati@egguser-virtual-machine_dpi_25524/linux_gcc-5.3.0/vsim_auto_compile.so"  "/tmp/andrea.bettati@egguser-virtual-machine_dpi_25524/linux_gcc-5.3.0/exportwrapper.o"   -Wl,-Bsymbolic '
# (vsim-50) A call to system(/opt/questasim/gcc-5.3.0-linux/bin/gcc -shared -fPIC -m32 -B/usr/lib32 -o "/tmp/andrea.bettati@egguser-virtual-machine_dpi_25524/linux_gcc-5.3.0/vsim_auto_compile.so"  "/tmp/andrea.bettati@egguser-virtual-machine_dpi_25524/linux_gcc-5.3.0/exportwrapper.o"   -Wl,-Bsymbolic  >'/tmp/questatmp.hE2MSZ' 2>&1) returned error code '1'.
# The logfile contains the following messages:
# /opt/questasim/gcc-5.3.0-linux/bin/../libexec/gcc/i686-pc-linux-gnu/5.3.0/ld: /usr/lib32/crti.o: unrecognized relocation (0x2b) in section `.init'
# /opt/questasim/gcc-5.3.0-linux/bin/../libexec/gcc/i686-pc-linux-gnu/5.3.0/ld: final link failed: Bad value
# collect2: error: ld returned 1 exit status
# 
# No such file or directory. (errno = ENOENT)
# 
# 
# FATAL ERROR while loading design
# Error loading design
Error loading design
# End time: 10:08:13 on Sep 02,2020, Elapsed time: 0:00:01
# Errors: 1, Warnings: 0
Makefile:127: recipe for target 'vsim-run' failed
make: *** [vsim-run] Error 12
```

Any hints on what could be the problem?
I think that [this issue](https://github.com/pulp-platform/pulpino/issues/110) could be related, but I can't find a cv32e40p equivalent of the `vsim/tcl_files/config/vsim.tcl` mentioned in the solution.

abettati, 2020-09-02:
From my understanding, the issue should be related to a mismatch betweend the GCC installed on my server (ggc-7.5.0) which is used to compile lib32, hence to create /usr/lib32/crti.o and the GCC version used by QuestaSim for the linking:
`/opt/questasim/gcc-5.3.0-linux/bin/../libexec/gcc/i686-pc-linux-gnu/5.3.0/ld`

For completeness, I run on:
kernel 5.4.0
ubuntu 18.04

abettati, 2020-09-02:
OK I think I resolved the issue by modifying the `modelsim.ini` file in my QuestaSim installation folder (`/opt/questasim` in my case).

In the `modelsim.ini` file I added the path to GCC installation:
```
; Specify gcc compiler used in the compilation of automatically generated DPI exportwrapper.
; Use custom gcc compiler located at this path rather than the default path.
; The path should point directly at a compiler executable.
DpiCppPath = /usr/bin/gcc
```

Silabs-ArjanB, 2020-09-03:
Hi @abettati Thank you for the update. Could you please close the issue as it seems you resolved it yourself?

closed 2020-09-03 by abettati