openhwfoundation/cv32e40p #481: Cause bits in dcsr should be read only

silabs-oysteink, 2020-09-04:
According to the riscv debug specification v0.13.2, the cause bits ([8:6]) in dcsr should be read only. During debug testing, I found that code executing in debug mode can actually write to these bits, causing a mismatch with the reference model.

eroom1966, 2020-09-09:
Documentation still shows this field as R/W - should this be recorded elsewhere ?

Silabs-ArjanB, 2020-09-09:
Hi @eroom1966 Could you please file a bug for that on https://github.com/openhwgroup/core-v-docs/issues?

silabs-oysteink, 2020-09-09:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Hi,

There is already an issue for the dcsr documentation, you could just update that one 😊
https://github.com/openhwgroup/core-v-docs/issues/163

Øystein