openhwfoundation/cv32e40p #484: cv32e40p_register_file_test_wrap has unused BIST ports

MikeOpenHWGroup, 2020-09-04:
Module `cv32e40p_register_file_test_wrap` has a set of un-used BIST ports.  It is instantiated in module `cv32e40p_id_stage`:

```
cv32e40p_register_file_test_wrap
  #(
    .ADDR_WIDTH(6),
    .FPU(FPU),
    .PULP_ZFINX(PULP_ZFINX)
  )
  registers_i
  (
    .clk                ( clk                ),
    .rst_n              ( rst_n              ),

// sniped port-map....

     // BIST ENABLE
     .BIST        ( 1'b0                  ), // PLEASE CONNECT ME;

     // BIST ports
     .CSN_T       ( 1'b0                ), // PLEASE CONNECT ME; Synthesis will remove me if unconnected
     .WEN_T       ( 1'b0                ), // PLEASE CONNECT ME; Synthesis will remove me if unconnected
     .A_T         ( 6'b0                ), // PLEASE CONNECT ME; Synthesis will remove me if unconnected
     .D_T         (32'b0                ), // PLEASE CONNECT ME; Synthesis will remove me if unconnected
     .Q_T         (                     )
  );
```

As far as I am aware, this is the only BIST interface in the core and the test_wrap is the only module that provides (or needs) BIST access.  Also, BIST is not mentioned in the User Manual.   I recommend that we either:

1. Remove the BIST logic and ports altogether, or,
2. Bring these ports to the top-level of the core and document their use in the User Manual.

This issue was found by looking at holes in [code coverage](https://openhwgroup.metrics.ca/cv32e40p_verif_3-core-v-verif/coverage/code/regressionRuns/cc3c865f-5cee-4faa-a9de-dff8333f55c0) from a Compliance test-suite regression run on the 6fbd88c645d2b51c316af6eda79bab3e4c284093 hash of this repo.

Silabs-ArjanB, 2020-09-06:
Hi @davideschiavone , what do you think about this. For me option 2 is not really an option. Either we leave it like this or (preferred by us) we remove this BIST related logic all together (it is not needed as far as I am concerned)

davideschiavone, 2020-09-21:
Let me ask the guy that made that change in the RF. Coming back soon

Silabs-ArjanB, 2020-09-21:
Hi @igorloi We all seem to agree that bringing these signals to the top is not the way to go. Also we do not want ifdefs in the code, so for me the only two options are to 'leave it as is' or 'remove the wrapper all together'.

MikeOpenHWGroup, 2020-09-21:
@Silabs-ArjanB said:
> the only two options are to 'leave it as is' or 'remove the wrapper all together'

I agree with this, and would ask if we decide on 'leave as is' that the existence of this interface be documented in the User Manual, including specific instructions on what the user should do if they do not want to use BIST ("do nothing" in this case).

We could also consider adding code coverage exclusions attributes in the RTL, but these may be tool vendor-specific.

davideschiavone, 2020-09-21:
I vote for removing it

MikeOpenHWGroup, 2020-09-21:
@davideschiavone disse:

> I vote for removing it

Even better.

Silabs-ArjanB, 2020-09-22:
Perfect, I will remove it

Silabs-ArjanB, 2020-09-24:
Hi @MikeOpenHWGroup Can this issue be closed now?

MikeOpenHWGroup, 2020-09-24:
Why yes it can.  :wink:  Thanks for this.

closed 2020-09-24 by MikeOpenHWGroup