openhwfoundation/cv32e40p #485: Unique case violations at t=0

MikeOpenHWGroup, 2020-09-04:
When running a simulation on the 6fbd88c645d2b51c316af6eda79bab3e4c284093 hash of the RTL, **XRUN** reports the following unique case violations at t=0:

```
xmsim: *W,NOCOND: Unique case violation:  Every case item expression was false.
            File: /home/v51366/mydata/GitHubRepos/openhwgroup/core-v-verif/master/core-v-cores/cv32e40p/rtl/cv32e40p_cs_registers.sv, line = 1094, pos = 14
           Scope: uvmt_cv32_tb.dut_wrap.cv32e40p_wrapper_i.core_i.cs_registers_i
            Time: 0 FS + 72

xmsim: *W,NOCOND: Unique case violation:  Every case item expression was false.
            File: /home/v51366/mydata/GitHubRepos/openhwgroup/core-v-verif/master/core-v-cores/cv32e40p/rtl/cv32e40p_prefetch_controller.sv, line = 168, pos = 14
           Scope: uvmt_cv32_tb.dut_wrap.cv32e40p_wrapper_i.core_i.if_stage_i.prefetch_buffer_i.prefetch_controller_i
            Time: 0 FS + 72

xmsim: *W,NOCOND: Unique case violation:  Every case item expression was false.
            File: /home/v51366/mydata/GitHubRepos/openhwgroup/core-v-verif/master/core-v-cores/cv32e40p/rtl/cv32e40p_obi_interface.sv, line = 131, pos = 16
           Scope: uvmt_cv32_tb.dut_wrap.cv32e40p_wrapper_i.core_i.if_stage_i.prefetch_buffer_i.instruction_obi_i.genblk1
            Time: 0 FS + 72
```
Note that there are no other errors in the simulation.

The same violations are reported by **DSIM**.   I have confirmed that the violations go away if the `unique case` is replaced with `unique0 case`.

### Steps to Reproduce
```
$ git clone https://github.com/openhwgroup/core-v-verif
$ cd core-v-verif/cv32/sim/uvmt_cv32
$ make SIMULATOR=xrun sanity
```

Silabs-ArjanB, 2020-09-06:
This looks like an error of core-v-verif itself. Specifically not all inputs to cv32e40p_core are assigned at time 0 and some are not driven at at any time. All input signals need to be valid at all times, especially the clock input (which is not the case).

[image]

@MikeOpenHWGroup Can you fix the environment in core-v-verif itself (I cannot move this issue to the correct repos, so maybe you can duplicate it there and close this one).

MikeOpenHWGroup, 2020-09-06:
Hi @Silabs-ArjanB, you are almost certainly correct.   We recently made some changes to the way clk and rst are driven (see https://github.com/openhwgroup/core-v-verif/pull/214) and this is probably the root cause.  I have opened core-v-verif issue [#219](https://github.com/openhwgroup/core-v-verif/issues/219) to ensure this is investigated.

However, I do not agree with your statement:
> All input signals need to be valid at all times, especially the clock input

My thoughts on this are:
1. Both clock and reset can be 'bX from t=0 to some t>0
2. Once clk or reset are driven to a valid value, either 'b0 or 'b1, the testbench must ensure they are valid until the end of the simulation.
3. If the DUT is not allowed to sample a specific input, the testbench may drive 'bX on that input.  For example when `instr_rvalid_i` is 1'b0, then `instr_rdata_i` may be driven X.

MikeOpenHWGroup, 2020-09-06:
Update.   I conducted a couple of experiments:

1. Drive `clk_i` to 'b0 starting at t=0.  This had no effect and the unique case violations were still reported.
2. Drive `clk_i` to 'b0 and `rst_ni` to 'b0 starting at t=0.  This clears up the unique case violations.

I believe (did not check) that the reason asserting reset at t=0 (rather than t>0) "fixes" the problem is because the debug and interrupt agents are reset by the same signal and both drive 'X until reset.

IMHO this is a simulation artifact that will probably not affect the real gates (after all, once reset happens, everything settles down).  Having said that, I think should look at fixing it in the RTL.

Silabs-ArjanB, 2020-09-07:
>     3\. If the DUT is not allowed to sample a specific input, the testbench may drive 'bX on that input.  For example when `instr_rvalid_i` is 1'b0, then `instr_rdata_i` may be driven X.

If you drive it X on instr_rdata_i when instr_rvalid_i = 0, then (especially for netlist simulations) we are not guaranteeing anything with respect to assertions firing, etc. (as you experienced). Digital is allowed to sample all inputs on every clock edge, no matter what 'valid' signals are indicating (but of course from a functional point of view the value of some inputs will not matter when the relevant valid signal is low). Driving X or leaving inputs unconnected is asking for problems in general during simulation.

Silabs-ArjanB, 2020-09-07:
At first I thought this was due to undriven inputs in the testbench (which I still think are a bad idea), but now I see that we are using cases statements in always_comb blocks with the case depending on a flip-flop (which could perfectly well be X at time 0):

always_comb
begin
unique case(flop_q)
// Default (transparent) state. Transaction requests are passed directly onto the OBI A channel.
1'b0:
begin
....
end
1'b1:
begin
...
end
endcase
end

Originally we had a default case alternative here (which led to Lint issueing warnings). After the Lint fix we now have simulator warnings.

MikeOpenHWGroup, 2020-09-07:
Ah linters.  Can't live without 'em, can't live with them either.   A couple of comments:

> we are using cases statements in always_comb blocks with the case depending on a flip-flop

What if we used a `case` statement, rather than `unique case`?  In the above snippet, unique case has not value because all possible (2-state) values of flop_q are defined.

> Driving X or leaving inputs unconnected is asking for problems in general during simulation.

I disagree.  Deliberately driving X is a valid technique for finding out if the RTL is sensitive to invalid inputs.  Of course, this needs to be disabled for gate sims, and should be avoided in assertions.

MikeOpenHWGroup, 2020-09-07:
Fair enough @Silabs-ArjanB .  There is no "perfect" answer, but there are several viable choices.   I have no strong preferences as long as we are all aware.   Whatever you, @silabs-PaulZ and @davideschiavone agree to is fine by me.

Silabs-ArjanB, 2020-09-08:
Paul, Steve and I talked it through and we decided to change the RTL, see https://github.com/openhwgroup/cv32e40p/pull/492

We do _not_ want to mandate reset assertion at t=0.

MikeOpenHWGroup, 2020-09-08:
Nice!   I cloned the latest RTL and the unique case violations are all gone.  Thanks!

closed 2020-09-08 by MikeOpenHWGroup