openhwfoundation/cv32e40p #486: Property in cv32e40p_prefetch_controller_sva.sv constantly fires

MikeOpenHWGroup, 2020-09-04:
SV property `p_branch_invalidates_fifo` in cv32e40p_prefetch_controller_sva.sv on hash 6fbd88c645d2b51c316af6eda79bab3e4c284093 constantly fires.  It looks like there is a hold-over from a "unit test":
```
  // Check that after a taken branch the initial FIFO output is not accepted
  property p_branch_invalidates_fifo;
     @(posedge clk) disable iff (!rst_n) (branch_i || 1) |-> (!(fetch_valid_o && fetch_ready_i));
  endproperty
```

The antecendent says **(branch_i || 1)**, which just _can't_ be what was intended.   I am pretty sure that we want is:
```
@(posedge clk) disable iff (!rst_n) (branch_i) |-> (!(fetch_valid_o && fetch_ready_i));
```
Note that I have not checked-in the ability to run this with **XRUN** in the UVM environment (coming soon).  Let me know if you need that in order to resolve this issue.

Silabs-ArjanB, 2020-09-06:
Hi @MikeOpenHWGroup This was indeed not intended (was test code), my bad. Will fix it right away.

Silabs-ArjanB, 2020-09-06:
Fixed with https://github.com/openhwgroup/cv32e40p/pull/488

Silabs-ArjanB, 2020-09-14:
Hi @MikeOpenHWGroup Can this be closed now?

MikeOpenHWGroup, 2020-09-14:
Yes, it can.  Thanks for the gentle reminder.

closed 2020-09-14 by MikeOpenHWGroup