openhwfoundation/cv32e40p #489: Debug Mode : ebreak instruction does not update dpc, dcsr before retire

eroom1966, 2020-09-08:
when transitioning into Debug Mode, from Machine mode due to an ebreak instruction.
At the point where the ebreak is retured, the dpc and dcsr registers do not contain the correct value.
The values are correct 1 cycle after the instruction has retired.
This causes a miscompare as these registers are compared at the ebreak retire

(FYI)
@MikeOpenHWGroup 
@silabs-oysteink

MikeOpenHWGroup, 2020-09-08:
Thanks for opening the issue @eroom1966.  Can you provide some specific steps to recreate this issue?  For example, the command-line for running the test that illustrates the error.

Silabs-ArjanB, 2020-09-08:
Hi @eroom1966 Is the CV32E40P ever in an invalid state or is this purely about trying to align the ISS state with the RTL state (which could be potentially solved in the behavioral tracer or by ignoring the step-by-step comparison for a cycle)

eroom1966, 2020-09-08:
@Silabs-ArjanB 
I think the internal state of the CV32E40P is correct.
I think the issue is only in the tracer/retire code which has indicated a retire one cycle too early, hence the state of dpc and dcsr is not correct for comparison purposes to the RM
The miscompare only exists for one instruction.

Silabs-ArjanB, 2020-09-08:
Hi @eroom1966 Would you be okay with suppressing the check for 1 cycle (knowing that if there was an actual issue the following cycles would cause a mismatch as well)? I prefer not to change the RTL if not really needed.

eroom1966, 2020-09-08:
@Silabs-ArjanB 
I would consider this a workaround rather than a solution - I would defer to @MikeOpenHWGroup to decide. Personally I would prefer the tracer to be fixed at source, workarounds are good, but tend to ignore the many potential side effects that may occur.

MikeOpenHWGroup, 2020-09-08:
AFAIK, the tracer is not RTL, so I do not understand the concern about changing it.

eroom1966, 2020-09-08:
@GTumbush @davideschiavone 
Hi Greg, Davide,
I wonder if you could help with the above, I note that @davideschiavone made changes to the retire behavior for the uret and mret instructions in order to ensure that the register state was correct at retirement.
It would appear that this is a similar issue with the ebreak instruction as described above the dpc and dcsr do not contain the correct values until a cycle later, so it would need changing thus

`if(~(trace.str == "mret" || trace.str == "uret")) begin`
to
`if(~(trace.str == "mret" || trace.str == "uret" || trace.str == "ebreak")) begin`

Does this look correct ?
Thx
Lee

eroom1966, 2020-09-09:
addendum, to get this to work I had to do the following
if(~(trace.str == "mret" || trace.str == "uret" || trace.str == "ebreak" || trace.str == "c.ebreak")) begin

GTumbush, 2020-09-09:
Hi Lee,
   Unfortunately, EM has not used the ISS in debug mode.  It seems the ISS is in a state in which we should.  I see no issues with your updates to the tracer but it would be best if someone from the design team blessed the change.

Silabs-ArjanB, 2020-09-17:
Hi @eroom1966 This issue should be fixed now; can you close it if agreed? (core-v-verif will be made to point to the latest RTL)

eroom1966, 2020-09-17:
agreed - closing

closed 2020-09-17 by eroom1966