openhwfoundation/cv32e40p #490: Debug Mode : dpc contains wrong value after dret to trigger match address==data2

eroom1966, 2020-09-08:
if a dret occurs from Debug mode into Machine mode, and the dpc address is the same as the tdata2 trigger data. The match is true and the core goes back into Debug mode, but the dpc is incorrectly set to the instruction following the address match (which did not execute)

According to the spec, dpc should contain the address of the next instruction to be executed. As the instruction at address == tdata2 did not execute, then the dpc should be the same as the value in tdata2

(FYI)
@MikeOpenHWGroup 
@silabs-oysteink

MikeOpenHWGroup, 2020-09-08:
Thanks for opening the issue @eroom1966.  Can you provide some specific steps to recreate this issue?  For example, the command-line for running the test that illustrates the error.

eroom1966, 2020-09-08:
Its quite complicated, but If I run @silabs-oysteink  debug_test, I can see the issue.
I am not sure if this test exists anywhere other than in Oystein's fork/branch

The dpc definitely holds the wrong value after trying to execute an instruction whose address matches the tdata2 trigger. this could be related to the fact that this is the first instruction executed after the dret from Debug Mode. 
I observe dret is executed, transitioning from Debug to Machine Moide, the dpc holds the same address for the tdata2 trigger address, so an immediate transition goes back from Machine to Debug. At this point I would expect the dpc to hold the same value, but it holds the next instruction address.

silabs-oysteink, 2020-09-08:
The debug_test also exist on the master. The same behavior should be visible there as well as the differences in the test itself between master and my fork/branch is minimal.

MikeOpenHWGroup, 2020-09-08:
@eroom1966 and @silabs-oysteink, please include specific instructions for reproducing the error in this issue report.  We cannot expect the designers to solve the issue unless they can see the issue.  I will typically include specific command-line instructions, including the command-line(s) required to clone the specific repo/fork/branch required.

silabs-oysteink, 2020-09-08:
1: Check out core-v-verif master from openhw github
2: For me it does not compile with assertions, so I changed line 56 in cv32/sim/uvm_cv32/Makefile to "SV_CMP_FLAGS ?= #"+define+CV32E40P_ASSERT_ON"
"
3:  Run "make debug_test USE_ISS=NO WAVES=YES" to run test. Excluded ISS as master version explodes in mismatches.

After complesion, you can observe the wrong dpc at time 783,963.3 ns

silabs-PaulZ, 2020-09-09:
Thanks @eroom1966 , @silabs-oysteink , @MikeOpenHWGroup ,

I see the issue and will look into the fix.
[image]

The jist of the problem is that the PC of the fetched instruction is being saved for single step, regardless of a trigger match (which should be the PC of the decoded instruction).

silabs-PaulZ, 2020-09-10:
PR #495 fixes DPC stored valued due to trigger.

[image]

eroom1966, 2020-09-10:
@silabs-PaulZ 
Hi Paul, great news @silabs-oysteink and I will test tomorrow 

I also wanted to give you a heads up, we are seeing DPC set one cycle early when executing in single step. This could just be a tracer issue, but we see the dpc updated with the address of the instruction after the single step, upon execution of the preceding instruction
More info to follow hopefully 

Thx
Lee

Silabs-ArjanB, 2020-09-24:
Hi @eroom1966 Can you please close this issue if you agree it has been fixed or let us know what is still broken?

eroom1966, 2020-09-24:
yes this is fixed - closing

closed 2020-09-24 by eroom1966