openhwfoundation/cv32e40p #496: The step or resume operation is error(depc error) when the SW/HW bp is set in a J instruction

apollojason, 2020-09-11:
Hi Guys：
I use the cv32e40p core running a simple test C code.
I set a SW bp before a “jump” instruction, when the cpu fetched the SW bp instruction, the cpu is halted.
But after that, i use the “step" cmd in telnet, the PC in CPU is wrong, it's not the correct PC which the JUMP instruction expected , the depc has been changed.
The same case is happened,when i set a HW bp in cv32e40p.
If i remove the SW bp or HW bp, when the CPU matched the breakpoint，the step or resume function is OK.
The debug module, used from the PULP riscv-dbg, and the debug environment is openocd and tilenet.
BTW，the same question is happened in IBEX, but when the CPU matched the HW bp and using the "step or resume"function funtion, the depc in IBEX is correct. 

So, i think this is a bug
please see the following figure

[image]
[image]
[image]
[image]

Silabs-ArjanB, 2020-09-11:
Hi @apollojason Thank you very much for your detailed bug report. Could you please let us know what git hash you used when experiencing above issue? Yesterday we made an depc related fix (https://github.com/openhwgroup/cv32e40p/pull/495) , so if you are not using cf064286636d6ff68660f666d1213bef5b7d2883 could you please let us know if the issue is still present in that version as well?

apollojason, 2020-09-11:
@Silabs-ArjanB 
Hi ArjanB：
      I cloned the cv32e40p RTL code in 08/31/2020, master branch.
      the commit ID is e1d3d780616e0d5b6ed1fa7b89b3cb038bf8b87f 
      Did u check the HW bp function, when the cpu matched the HW bp? Because i have tested the ibex core, when the ibex core has been matched the HW bp in trigger, the step or resume function is corrected, but the SW bp function in ibex is wrong, this error is same with the cv32e40p SW bp function error.

Silabs-ArjanB, 2020-09-11:
Hi @apollojason , we did quite some fixes to the CV32E40P in the last week, so it would really help us if you could repeat your check on the latest commit to the master branch. 

@silabs-oysteink , are you also specifically verifying the scenario described by @apollojason ?

apollojason, 2020-09-11:
@Silabs-ArjanB 
Hi ArjanB:
    OK, i'll update the design and checking again. 
    Hoping to provide assistance to you!

Silabs-ArjanB, 2020-09-11:
@apollojason Thank you very much! I realize it is yet extra work, but it really helps us

silabs-oysteink, 2020-09-11:
@Silabs-ArjanB I'm don't have a test that implements the exact specifics of what @apollojason describes. But we did see that the dpc was wrong then single stepping and the trigger matched an address. This issue is fixed in PR #495. We are also currently debugging a similiar issue where the dpc seems to be wrong when single stepping and entering exception routines - not sure if it is RTL or tracer that causes that one yet.

apollojason, 2020-09-11:
@Silabs-ArjanB @silabs-oysteink 
Hi：
    I have got the new version CPU and finished the debug SW breakpoint test.
    There's a good news. It seems like the cpu is good running. 
    Could u double check the wave? I also used the same opertaion flow for testing the SW bp.

[image]

Silabs-ArjanB, 2020-09-11:
Hi @apollojason Thank you for running this again on the latest RTL. I will ask Paul to have a look at this as he knows the details about this part of the RTL; @silabs-PaulZ could you please check the lat waveform and indicate if this (now) looks okay or not?

silabs-PaulZ, 2020-09-11:
Hi @apollojason , Thank you for reporting this. As mentioned before, we recently fixed this issue.

I can confirm that looking by at your waveforms the fix has been applied. Before the fix, the machine would incorrectly go to the DBG_TAKEN_IF state (e.g. enter debug mode and store the PC of the instruction fetched (if)) which would improperly store the next instruction. The fix now makes the machine go to the DBG_TAKEN_ID state (e.g. enter debug mode and store the PC of the instruction being decoded (id))...as this instruction was decoded but not executed.

Before closing the ticket can you briefly explain the difference in your setup between a HW breakpoint and SW breakpoint. 
I would assume by HW breakpoint that you are using a trigger (TDATA1/MCONTROL & TDATA2) match which would cause the processor to always jump into debug mode (in CV32E40P).
I would assume an SW breakpoint as using the ebreak instruction (that either enters debug or executes software exception routine).

apollojason, 2020-09-12:
@silabs-PaulZ 
Hi Paul：
     Thank you for explaining the theory of the "depc update mechanism".  I couldn‘t agree more.
     About the difference between the HW bp and SW bp,  they are same from a debugging perspective.
     When a debuger user the OpenOCD set a bp, if the bp is a HW breakpoint, it will user the trigger in the cpu core, if the bp is a SW breakpoint, the OpenOCD will read the instruction from the memory(breakpoint PC), and rewrite the EBREAK instruction to the memory location.
     If the instruction is stored in the flash ram, the only way for setting bp is HW breakpoint.

Silabs-ArjanB, 2020-09-12:
Hi @apollojason It seems this issue has been solved. If you agree can you please close this issue (if not, please let us know why not)? Thank you for your help!

apollojason, 2020-09-12:
@Silabs-ArjanB @silabs-PaulZ 
Hi：
      I agree you. I'll close this issue.
      Do you have a constraint file for synthesize？ I want to have a try for DC.

Silabs-ArjanB, 2020-09-12:
Hi @Silabs-ArjanB We have constraints, but it will take some time to release them.

apollojason, 2020-09-12:
@Silabs-ArjanB 
 Hi：
     OK，thank you very much!  I wish the cv32e40p cpu will be tipped out successfully.

Silabs-ArjanB, 2020-10-28:
Re-opening and closing because somehow github is reporting this issue as still open

closed 2020-10-28 by Silabs-ArjanB