openhwfoundation/cv32e40p #497: Questions about Exceptions in CV32E40P

MikeOpenHWGroup, 2020-09-11:
During the review of the CV32E40P Exceptions DV Plan the following questions came up:

1. Is there an "Illegal Memory Access" exception?   That is, what happens when the core attempts to fetch and instruction or load/store data from/to an address that does not exist?   It seems that the core simply hangs, is that the case?
2. Does the CV32E40P support any ALU instructions?   For example, what happens on a divide-by-zero?

Silabs-ArjanB, 2020-09-12:
1A) Is there an "Illegal Memory Access" exception? No, CV32E40P does not use these exceptions. There are no bus error pins; this feature is scheduled to be added on CV32E40.
1B) That is, what happens when the core attempts to fetch and instruction or load/store data from/to an address that does not exist? That depends on the environment; it could for example issue an interrupt if this happens; there is no way to signal that this happens via the OBI err pin as CV32E40P does not implement that pin.
1C) It seems that the core simply hangs, is that the case? No, this must be a test  environment artefact, the CV32E40P is not aware which addresses do or do not exist, it always expects a valid response from the environment.

2A) Does the CV32E40P support any ALU instructions? Yes, ADD, SUB, AND, etc.. They never cause exceptions.
2B) For example, what happens on a divide-by-zero? See table 7.1 of the RISC-V unprivileged spec. (No exceptions will occur)

MikeOpenHWGroup, 2020-09-13:
Thanks for the clarification @Silabs-ArjanB, I'll close this issue now.

About my second question:
> Does the CV32E40P support any ALU instructions?

That was an embarassing typo.  I meant ALU **_exceptions_**, not ALU _instructions_.  Its pretty obvious processor core should support instructions.  :laughing:

closed 2020-09-13 by MikeOpenHWGroup