openhwfoundation/cv32e40p #498: Wrong result from lw instruction

joestoy, 2020-09-11:
## Bug Title
Wrong load value return from lw instruction

### Component
load-store unit
**Component:RTL**: For issues in the RTL (e.g. for files in the rtl directory)

### Steps to Reproduce
Please provide:
1. git hash: 1607d8b675864db1aa013364fd4444a665331830
2. Command line: to run coremark.exe in verilator simulation,
    ./exe_cv32e40p_sim  +v1 +tohost +jtag_port=6600 +vpi_port=7700
with Mem.hex in the working directory.  (Note: the port numbers are used only with various gdb connections, which you will not need.)

This issue concerns unexpected (by me) behavior in the load-store unit.  I am using the above commit  of cv32e40p.  I attach waves from a verilator simulation, the elf file and disassembled dump of the program being run, and a screenshot to help description of the issue, and the simulator executable for ease of reproducing the problem (let me know if there are any difficulties).

The relevant instruction is at line 916 of the dump file, at address 0xc0000a7c: lw a5,0(a5).  In the screenshot, the top group shows the (compressed) instruction arriving at 9353686; the next little group shows the instruction correctly decompressed (0x0007A783) at 9353696.  The third group shows the memory read request (to address 0xC00EF800) at 9353706, and the correct reply (0xC00EF8E8) at 9353926.  The next group shows activity in the register file, which is as expected until 9353936, where the destination register (address 0x0F) gets 0xE8000000 instead of the expected 0xC00EF8E8.

The final group shows activity in the load_store unit.  You see that data_rdata_i gets the correct value.  But data_rdata_ext seems to be shifting it in from the more significant end (I haven't yet worked out why), and data_rdata_ex_o seems to be sampling it before it is completely there.

I hope this gives you enough to analyze the problem, but don't hesitate to ask if you would like me to try to get more.

joe stoy

[exe_cv32e40p_sim.gz](https://github.com/openhwgroup/cv32e40p/files/5211744/exe_cv32e40p_sim.gz)
[Mem.hex.gz](https://github.com/openhwgroup/cv32e40p/files/5211748/Mem.hex.gz)
[coremark.exe.gz](https://github.com/openhwgroup/cv32e40p/files/5211704/coremark.exe.gz)
[coremark.dump.gz](https://github.com/openhwgroup/cv32e40p/files/5211705/coremark.dump.gz)
[vlt_dump.fst.gz](https://github.com/openhwgroup/cv32e40p/files/5211709/vlt_dump.fst.gz)
[image]

Silabs-ArjanB, 2020-09-12:
Hi @joestoy Thank you for submitting this detailed issue report. It looks really weird to me. The data gets wrongly aligned because the core thinks this is a misaligned load (which it is not). In the waveform below rdata_offset_q should remain 0 at/after the clock edge marked '2' in the figure below. I do not use Verilator myself, but at first glance this looks like a simulator issue. Can you explain why rdata_offset_q goes to 1 at the circled point in below waveform (that seems the root cause of the issue). If you look at the RTL, then you will see that rdata_offset_q can only change on rising clock edges for which ctrl_update = 1, and ctrl_update =  lsu_ready_ex_o && data_req_ex_i, so as shown in the waveform ctrl_update should be 1 at point '1' and not at point '2'. Also I noticed that the wave file contains CS and NS, which do not exist in the load/store unit file; can you explain that as well? You are also running on a quite old version of the core (but I don't think that causes this issue).

[image]

```
  // FF for rdata alignment and sign-extension
  always_ff @(posedge clk, negedge rst_n)
  begin
    if(rst_n == 1'b0)
    begin
      data_type_q       <= '0;
      rdata_offset_q    <= '0;
      data_sign_ext_q   <= '0;
      data_we_q         <= 1'b0;
      data_load_event_q <= 1'b0;
    end
    else if (ctrl_update) // request was granted, we wait for rvalid and can continue to WB
    begin
      data_type_q       <= data_type_ex_i;
      rdata_offset_q    <= data_addr_int[1:0];
      data_sign_ext_q   <= data_sign_ext_ex_i;
      data_we_q         <= data_we_ex_i;
      data_load_event_q <= data_load_event_ex_i;
    end
  end

  assign ctrl_update = lsu_ready_ex_o && data_req_ex_i;
```

MikeOpenHWGroup, 2020-09-13:
@Silabs-ArjanB said:

>  I do not use Verilator myself, but at first glance this looks like a simulator issue.

I would be willing to help out here as I have access to Verilator, Cadence Xcelium and Metrics dsim simulators.   @joestoy, can you provide more details to run the verilator test?  I do not know what "coremark.exe" refers to.  A set of command-lines starting with the `git clone` command would allow me to reproduce the error you are seeing and then attempt to reproduce the same error using a different simulator.

> You are also running on a quite old version of the core

Indeed.  This hash is almost two months old and a lot of significant changes have happened since then.

Silabs-ArjanB, 2020-09-16:
Hi @joestoy Could you please let me know if you agree that this is a simulator issue as opposed to an RTL issue?

joestoy, 2020-09-16:
@Silabs-ArjanB: Sorry for delay -- we've been a bit distracted by yesterday's OpenHW conference.  Yes, I think you're right.  I made a slight change to the surrounding SoC (no change whatever to the CV32E40P RTL), and can no longer reproduce the issue.  What's more, the CS and NS signals are no longer there.  I'm guessing they mean "current state" and "next state" (Cs always trails one cycle behind NS), and indicates that something prompted verilator to analyze the situation more carefully (and wrongly).  (Incidentally, some signals in those waves come one unit later than the clock edgs.  That's a (configurable) artefact of the bsc compiler when compiling BSV to Verilog, intended to be a notional reminder of propagation delays.  That could have been an incentive for verilator to be more careful -- though it doesn't usually do so.).  Thanks for your analysis, and sorry it seems to have been a red herring.

closed 2020-09-16 by joestoy

Silabs-ArjanB, 2020-09-16:
Hi @joestoy No problem at all; happy to see that the issue is gone (although it was not fully understood).