openhwfoundation/cv32e40p #500: Duplicate file in compilation | Tool-and-build

dawidzim, 2020-09-14:
Hey!
In `/bhv/cv32e40p_wrapper.sv` is:
```
`include "cv32e40p_core_log.sv"

`ifdef CV32E40P_APU_TRACE
  `include "cv32e40p_apu_tracer.sv"
`endif

`ifdef CV32E40P_TRACE_EXECUTION
  `include "cv32e40p_tracer.sv"
`endif
```
In my opinion we don't need to compile this files, so we should change line from `example_tb/core/Makefile`:
```
RTLSRC_BHV		:= $(wildcard $(RTLSRC_HOME)/bhv/*.sv)
```
to
```
RTLSRC_BHV		:= $(filter-out $(RTLSRC_HOME)/bhv/cv32e40p_tracer.sv $(RTLSRC_HOME)/bhv/cv32e40p_core_log.sv $(RTLSRC_HOME)/bhv/cv32e40p_apu_tracer.sv, $(wildcard $(RTLSRC_HOME)/bhv/*.sv))
```

Silabs-ArjanB, 2020-09-14:
Hi @dawidzim Agreed with that change. Could you please issue a pull request for that change?

dawidzim, 2020-09-14:
Hey @Silabs-ArjanB , sure PR #501

Silabs-ArjanB, 2020-09-14:
Hi @dawidzim Thank you; it has been merged now

dawidzim, 2020-09-14:
Ok, thanks:)

closed 2020-09-14 by dawidzim