openhwfoundation/cv32e40p #503: interrupt handler entry after CSRRC instruction clears local interrupt

strichmo, 2020-09-15:
It is not completely clear to me whether this is an architectural bug, or simply a scenario where I need to tune the ISS to.

In this case, the CSRs are being written to (including MIE and MSTATUS.MIE) while external interrupts are being generated randomly.  In this particular case an external interrupt [29] is being asserted one cycle before mie[29] is being disabled via a CSRRC write.  The interrupt controller sees the single cycle of enabled interrupt [29] and initiates its state machine to enter the interrupt.  The CSRRC completes and then the interrupt is jumped to (0x4a774).  The test fails because the ISS is seeing the updated MIE value and does not think [29] is enabled.  However that is separate testbench issue.

My question is architecturally does clearing an MIE signal a critical section for IRQ[29] (or perhaps in general for all interrupts for MSTATUS.MIE or any combinations of interrupts)?  

--If the pipeline were to execute one more instruction before taking the interrupt then there is no question there is an architectural violation.--

**Component:RTL**

### Steps to Reproduce
To reproduce, clone my fork at:
https://github.com/strichmo/core-v-verif/tree/strichmo/pr/rand_interrupt_wfi
Hash: 6446d0f0a02b1a491b75854cc5963743fa6caf67

% makecv32 comp_corev-dv gen_corev-dv test TEST=corev_rand_interrupt WAVES=1 SEED=204713540 COMP=1 CFG=no_pulp USER_RUN_FLAGS=+UVM_MAX_QUIT_COUNT=30,NO

strichmo, 2020-09-15:
[image]

Silabs-ArjanB, 2020-09-16:
Hi @strichmo Thank you for filing this issue. I looked at the wave and I consider it an actual RTL bug (but maybe for a different reason than you thought).

If you look at this code:

```
00000ee2 30453cf3 csrrc            x25, x10, 0x304 
00000ee6 0109a733 slt              x14, x19, x16   
```

then it should never be possible that 'slt' is the interrupted instruction because of irq_i[29]. So in below waveform the combination of mepc == 0xee6 (i.e. SLT instruction address) and irq_id_o == 29 is wrong in my opinion. It is possible that the SLT gets interrupted, but then it should be because of irq_i[28]. Before fixing this I first want to check if the same could have happened without the occurrence of irq_i[28]. A potential fix might be adding some flush cycles in the RTL around the CSR write to MIE (which we would like to avoid) or to change the assignment to irq_id_o.

[image]

Silabs-ArjanB, 2020-09-17:
mcause reflect an IRQ that is not allowed to be taken as well. Easiest fix (at the cost of latency) is to cause flushes upon mie writes.

[image]

Silabs-ArjanB, 2020-09-18:
Hi @strichmo Could you please use the (not yet merged) repos from https://github.com/openhwgroup/cv32e40p/pull/508 and run an (interrupt) regression on it? The problem from this github issue should have been fixed with that PR.

Silabs-ArjanB, 2020-09-20:
Hi @strichmo Could you please run an (interrupt) regression on https://github.com/openhwgroup/cv32e40p/pull/511 before we perform the merge? It should fix issue #503 (but a timing optimization still needs to be performed before we can merge it).

Silabs-ArjanB, 2020-09-24:
Hi @strichmo Can you please close this issue if you agree it has been fixed?

strichmo, 2020-09-27:
Closing issue.  Regressions with random interrupts are running clean.

closed 2020-09-27 by strichmo