openhwfoundation/cv32e40p #509: Core executes wrong instruction

tzwaenn, 2020-09-18:
## Bug Title
After some dmem stalls, a retracted interrupt, and an additional debug interrupt, and instruction that should have been flushed gets exeuted as first instruction after entering debug mode.

### Component

**Component:RTL**

### Steps to Reproduce
This is a corner case sequence of events leading to an instruction being executed although it shouldn't be. It was found with OneSpin 360 formal verification and the Processor Verirfication App while proving the update of the program counter.
After enabling interrupts (not shown in beginning of trace), an lw is exeucted and there is some incoming interrupt. There are some dmem stalls on the lw and the interrupts are retracted, so they are not taken after all, but a flag instr_valid_irq_flush_q is set - seemingly indicating that the already fetched instruction sw still needs execution. So far so good. However, now, there is an incoming debug interrupt, so debug mode is entered while keeping that flag high. It seems the core then, after enering debug mode at t##0.  thinks it still needs to exeucte the sw instruction still in decode - at least we see it producing a dmem access at t##1 with the expected address ('h20 as result of the previous lw + 136 = 'ha8.
[image]
VCD can be provided if needed.

Silabs-ArjanB, 2020-09-22:
Hi @tzwaenn Thank you for this detailed bug report, very cool that your assertions find this! I will dive in it tomorrow, could you please also provide the VCD

tzwaenn, 2020-09-22:
Hi Arjan,

I was not expecting that the mundane task of checking the PC increment of a memory instruction would have any fallout - but that's the beauty of formal. I had to zip the vcd as the system would not accept the .vcd extension.
[RV_chk.ops.RV32I.vcd.zip](https://github.com/openhwgroup/cv32e40p/files/5262988/RV_chk.ops.RV32I.vcd.zip)
Sven

Silabs-ArjanB, 2020-09-23:
Hi @tzwaenn A couple of days ago the interrupt controller RTL and related states in the controller were greatly simplified (see https://github.com/openhwgroup/cv32e40p/pull/511). Specifically the 'replay after interrupt speculation' part has been removed from the RTL and I think that this caused the issue you found. So, I expect that my RTL updates of earlier this week (accidentally) already fixed this issue. Could you please run your checks again on the latest RTL?

tzwaenn, 2020-09-24:
Hi Arjan,

I can confirm that instr_valid_irq_flush_q is now formally stuck at 0, hence this particular issue no longer exists. You might want to remove instr_valid_irq_flush_q from the RTL moving forward...

davideschiavone, 2020-09-24:
Hello and thanks @tzwaenn , in the last version of the core that signal should not be present at all indeed.

Silabs-ArjanB, 2020-09-24:
Hi @tzwaenn Perfect, thank you for your help. Could you please close this issue if you agree the issue has been solved?

closed 2020-09-24 by tzwaenn