openhwfoundation/cv32e40p #510: Unexpected illegal instruction reading mcounteren

MikeOpenHWGroup, 2020-09-18:
The test-program `core-v-verif/cv32/tests/programs/custom/requested_csr_por/requested_csr_por.c` simply reads a set of CSRs.  The test flags two issues:

1. Reading `mcounteren` produces an illegal instruction exception.  This is not expected.
2. The value of `mepc` is off by one address.

### Test-program info
Line 408 is the read of `mcounteren`:
```
 __asm__ volatile("csrr %0, 0x306" : "=r"(mcounteren_rval));    // Not currently modeled

  //if (mcounteren_rval != 0x0) {
  //  printf("ERROR: CSR MCOUNTEREN not 0x0!\n\n");
  //  ++err_cnt;
  //}

  /////////////////////////////////////////////////////////////////////////////
  // These are read last because there should not have been any events which
  // caused these CSRs to be incremented up to this point.
  __asm__ volatile("csrr %0, 0x340" : "=r"(mscratch_rval));
```

This produces the following ASM code:
```
  __asm__ volatile("csrr %0, 0x306" : "=r"(mcounteren_rval));    // Not currently modeled
     5ea:	30602773          	csrr	a4,0x306
  //}

  /////////////////////////////////////////////////////////////////////////////
  // These are read last because there should not have been any events which
  // caused these CSRs to be incremented up to this point.
  __asm__ volatile("csrr %0, 0x340" : "=r"(mscratch_rval));
     5ee:	34002c73          	csrr	s8,0x340
```
The offending instruction is at address 0x5ea, but `mepc` reads 0x5ee.

### Steps to Reproduce
```
git clone https://github.com/openhwgroup/core-v-verif
cd core-v-verif/cv32/sim/uvmt_cv32
make test SIMULATOR=<sim> USE_ISS=NO TEST=requested_csr_por
```
Where `sim` is one of xrun, vsim, vcs or dsim, depending on your simulator of choice.
Disabling the ISS merely simplifies the error messages produces by the simulation.  The failure is the same either way.

jm4rtin, 2020-09-18:
@MikeOpenHWGroup The specification states:
> Registers mcounteren and scounteren are WARL registers that must be implemented if U-mode and S-mode are implemented.

The CSR is only implemented in the core if user mode is enabled. The documentation omits the description of this register because it's user mode only. It is listed in the user mode table which gets stripped out. Despite removing it from the machine mode table when the HPM got merged, the CSR got added back during the cleanup.

Removed: https://github.com/openhwgroup/core-v-docs/commit/ea7d9d4c9a2155ed93d5aabcf8352d15e61bc6ac
Added back: https://github.com/openhwgroup/core-v-docs/commit/5fdffa125ee65849478eaa7f399e0a61ac8d84fd

MikeOpenHWGroup, 2020-09-19:
> Despite removing it from the machine mode table when the HPM got merged, the CSR got added back during the cleanup.

That is probably my bad from [core-v-docs #172](https://github.com/openhwgroup/core-v-docs/pull/172).  I'll open an issue in core-v-docs to get that fixed.

OK, so the the situation is now as follows:

1. Reading `mcounteren` produces an illegal instruction exception. This **_is_** expected.
2. The value of `mepc` is off by one address.

I can close this issue and open a new one if desired.

Silabs-ArjanB, 2020-09-19:
Hi @MikeOpenHWGroup I cannot reproduce this issue. Could you please upload the generated files in core-v-verif/cv32/tests/programs/custom/requested_csr_por/ and a command line to run the test case such that these generated files are not overwritten? (Using mainline gcc the mepc is okay for me; using the PULP tool chain the test does not compile for me)

MikeOpenHWGroup, 2020-09-22:
Hi @Silabs-ArjanB.  I get more than a little worried when comments about toolchains pop up, so I took a seperate tack here:

- switched to the [OpenHW blessed toolchain](https://www.embecosm.com/resources/tool-chain-downloads/#corev).
- wrote a dedicated assembly testcase.
- Ensured the TB is pointed at the latest RTL and uses the proper top-level parameters, specifically PULP_XPULP=0

To reproduce, clone the issue510 branch of https://github.com/openhwgroup/core-v-verif and run (new) test-program `issue510`:
```
git clone -b issue510 https://github.com/openhwgroup/core-v-verif issue510
cd issue510/cv32/sim/uvmt_cv32
make test SIMULATOR=<sim> USE_ISS=NO TEST=issue510
```

The testcase is at `../../tests/programs/custom/issue510/`.  This is the simpliest possible test-program imaginable which basically boils down to `csrr x2, 0x306`, which is a read of mcounteren.  My results as follows:

1. Reading `mcounteren` produces an illegal instruction exception. This is expected.
2. The value of `mepc` now is as expected (good news).
3. The PC is cleared and the core endlessly attempts to fetch from 0x0 (unexpected!)

Silabs-ArjanB, 2020-09-22:
Hi @MikeOpenHWGroup The above code is much too simplified, basically there seems to be no exception handler and because MTVEC keeps the illegal instruction exception address at 0 (and the test environment returns 32'bx from that address= the shown tests stays in a loop (first encountering the illegal mcounteren instruction, then encountering the illegal 32'bx forever)

MikeOpenHWGroup, 2020-09-22:
Sorry @Silabs-ArjanB, maybe my last update had too many details and the following got lost in translation: when I use the COREV toolchain (the same one you pointed out above), the value of mepc is as expected.  So that is no longer the problem.

I'll investigate the exception handler issue and get back to you.

Silabs-ArjanB, 2020-09-22:
Hi @MikeOpenHWGroup I am not looking for a simpler test case, just a wave showing the original problem you mentioned. So as far as I am concerned you do not need to fix the exception handler issue as you already seem to have a failing test.

MikeOpenHWGroup, 2020-09-22:
I had to cut the test down to size since the compressed vcd was larger than 10M.   Here is the log and vcd from a reduced version of the test-program that illustrates the issue.

[dsim-requested_csr_por.log](https://github.com/openhwgroup/cv32e40p/files/5261991/dsim-requested_csr_por.log)
[dsim-requested_csr-por.vcd.gz](https://github.com/openhwgroup/cv32e40p/files/5261993/dsim-requested_csr-por.vcd.gz)

Silabs-ArjanB, 2020-09-22:
@MikeOpenHWGroup Thanks! I will look at it right away

Silabs-ArjanB, 2020-09-22:
Hi @MikeOpenHWGroup I am sorry, but I still do not see the wrong mepc as you mentioned at the start of this issue. The following is a screenshot from dsim-requested_csr-por.vcd.gz which looks fine to me.

[image]

MikeOpenHWGroup, 2020-09-22:
Interesting.  I see what you see, but when I subsequently read the `mepc`, it is off by one instructions (expecting 0x164, but read 0x168).   I've attached a log and waves for a slightly modified version of the test that reads the mepc.

[dsim-requested_csr_por_take2.log](https://github.com/openhwgroup/cv32e40p/files/5263029/dsim-requested_csr_por_take2.log)
[dsim-requested-csr-por_take2.vcd.gz](https://github.com/openhwgroup/cv32e40p/files/5263031/dsim-requested-csr-por_take2.vcd.gz)

Silabs-ArjanB, 2020-09-23:
Hi @MikeOpenHWGroup Thank you for the new waves. I now see that 0x168 is being read, but only after software first explicitly writes that value into mepc as shown with the 0x34129073 instruction at 0x2CE in below waveform:

[image]

In above waveform the first a read of mepc happens (correctly returning 0x164), then a (software based) write happens with value 0x168, and then another read (which now correctly returns value 0x168). So my question now is: Why is software writing 0x168 to mepc? (It looks like a test case error to me, but I can't see the source code)

MikeOpenHWGroup, 2020-09-23:
Hi @Silabs-ArjanB.  It appears you are right - I am using C macros to create assembly code and it is injecting unexpected writes.  Having said that, I am still not completely convinced that all is well, so I will assign this issue to myself and work to write a assembly test to either prove to myself everything is fine or definitively say that it is not.

Stay tuned and thanks for your patience.

Silabs-ArjanB, 2020-10-28:
Hi @MikeOpenHWGroup Is there still an issue here or has this been resolved by now?

MikeOpenHWGroup, 2020-10-28:
Yes, it has been resolved - toolchain issue.  Should have closed this a week ago.  Closing now.

closed 2020-10-28 by MikeOpenHWGroup