openhwfoundation/cv32e40p #529: Wrong dcsr.cause after sleep

silabs-oysteink, 2020-09-30:
## Wrong dcsr.cause after sleep
Cause bits in dcsr seems to wrong if the first instruction decoded after WFI + debug_req_i is an EBREAK.

1. Core is sleeping
2. debug_req_i is asserted
3. Core wakes up, fetches an ebreak instruction
4. Core enters debug mode, sets cause to 0x1 (ebreak)
5. Ebreak instruction does not retire

[image]

Debug spec v 0.13.2 states the following:

> If halt is requested while wfi is executing, then the hart must leave the stalled state, completing
this instruction’s execution, and then enter Debug Mode.

In this case, I would expect the core to disregard the ebreak and enter debug mode with cause=0x3 (external debugger). Instead it enters debug as if the ebreak caused it, although the ebreak does not retire.

### Component
**Component:RTL**

### Steps to Reproduce
1. Clone my fork of core-v-verif:  https://github.com/silabs-oysteink/core-v-verif.git
2. Check out branch silabs-oysteink_random-debug
3. from cv32/sim/uvmt_cv32:
4: make comp_corev-dv gen_corev-dv TEST=corev_rand_debug START_INDEX=0 NUM_TESTS=1
5: make test TEST=corev_rand_debug RUN_INDEX=0 WAVES=YES USE_ISS=NO

I've disabled the ISS as the current state of the repo requires an invalid license. Assertion will fire regardlessly.

Silabs-ArjanB, 2020-09-30:
@silabs-PaulZ I expect that exactly the same issue will happen when we try to assert debug_req_i upon cold start while the first regular program instruction would have been an ebreak.

silabs-PaulZ, 2020-09-30:
Thank you @silabs-oysteink for finding this. 

I submitted a fix with PR #531. This corner case occurs with WFI, reset, and also in general code execution.

Below are waveforms showing the new register ebrk_debug_entry_q that will capture if entry is due to a ebreak instruction. The top half of the waveform shows the scenario where debug_req preempts the ebreak instruction and cause is set to `haltreq`. The bottom half is where debug_req is after the taken ebreak instruction, asserting ebrk_debug_entry_q, and setting cause to `ebreak`.

[image]

Silabs-ArjanB, 2020-10-01:
Hi @silabs-oysteink Thank you for finding this issue. The related PR has been merged. Can you update core-v-verif to the latest CV32E40P repos and close this issue if you agree it has been fixed?

silabs-oysteink, 2020-10-01:
The issue is fixed, but another issue came up which seems related: If we get a debug_req_i while single stepping, we should enter debug mode with cause=3 (external request) instead of cause=4 (single step), as external debug request has higher priority than a single step.  This is captured by the same method as described above, only with updated RTL for this issue.

[image]

According to debug spec 0.13.2, the priority of cause in case of multiple events is as follows (from highest to lowest):

1. Trigger match
2. Ebreak
3. resethaltreq
4. haltreq
5. step

I'm not sure if I should make a new issue or continue on this one, as they seem very related.

silabs-PaulZ, 2020-10-02:
Hi @silabs-oysteink , I had originally made single step mask the debug_req in order to always honor the single step execution before returning back to debug mode. I see this doesn’t work well for all corner cases, such as in the case where the user keeps debug_req asserted when exiting debug mode. 

I’ve made another PR #538 which will fix this problem. 

In this wave, you can see that debug req occurs just before the instruction has an opportunity to retire and the cause = haltreq, with depc not changing.

 
[image]

In this example, you can see that debug req occurs just after the instruction is retired and the cause = step, with depc incrementing to next instruction.
 
[image]

Silabs-ArjanB, 2020-10-05:
Hi @silabs-oysteink https://github.com/openhwgroup/cv32e40p/pull/538 has been merged. If the above issue has been resolved can you please close this issue?

closed 2020-10-05 by silabs-oysteink