openhwfoundation/cv32e40p #533: Illegal Instruction Retires

shetalani, 2020-10-01:
### Issue Description
An illegal instruction increments the minstret counter as if it was a legal one.

### Component
**Component:RTL**

### RISC-V Specification

> 3.1.11: "The minstret CSR counts the number of instructions the hart has retired. The mcycle and minstret registers
have 64-bit precision on all RV32 and RV64 systems.

> The counter registers have an arbitrary value after system reset, and can be written with a given value. Any CSR write takes effect after the writing instruction has otherwise completed."

> The counter-inhibit register mcountinhibit is a 32-bit WARL register that controls which of the hardware performance-monitoring counters increment. The settings in this register only control whether the counters increment; their accessibility is not affected by the setting of this register.

> When the CY, IR, or HPMn bit in the mcountinhibit register is clear, the cycle, instret, or hpmcountern register increments as usual. When the CY, IR, or HPMn bit is set, the corresponding counter does not increment.

### Steps to Reproduce

As shown below, the following sequence of instructions happen: 
> csrrw x13, mcountinhibit, x18 -> csrrw x4, 0xa21, x13 -> fmadd.s f0, f0, f0, f0

The first instruction sets all the bits of mcountinhibit to '0 and thus allowing for the counters to be incremented. The last two instructions are illegal ones, for them machine csrs like mepc and mcause are updated storing their PCs and the cause of the exception, in this case the value 2. Looking at mhpmcounter value, its incremented by 1 for each of the two illegal instructions. The reason why it was not incremented when the first instruction got retired/ decoded is that the IR bit of mcountinhibit was set to 1. 

[image]

**Top Level Parameters**
```
cv32e40p_core #(
    .FPU                      ( 0 ),
    .NUM_MHPMCOUNTERS         ( 1 ),
    .PULP_CLUSTER             ( 0 ),
    .PULP_XPULP               ( 0 ),
    .PULP_ZFINX               ( 0 )
) 
```
---
Product: OneSpin 360 DV-Verify
App: Processor Verification App
Tool's version: 2020.2.0

silabs-PaulZ, 2020-10-02:
To expand on this, according to the following discussion, all traps should not increment the instret counter. This includes ebreak and ecall along with illegal instruction. 

https://lists.riscv.org/g/tech-base-isa-archive/message/78?p=,,,20,0,0,0::relevance,,instret,20,2,0,32208768
[image]

Silabs-ArjanB, 2020-10-02:
Hi @silabs-PaulZ Agreed with that; I will fix the fix

Silabs-ArjanB, 2020-10-06:
I will also change the 'compressed instruction count' into a 'retired compressed instruction count' (i.e. excluding the same categories as excluded for minstret

Silabs-ArjanB, 2020-10-12:
Hi @shetalani This issue should have been solved now. Could you please run the OneSpin checks again on the current master branch? Note that together with illegal instructions also ecall and ebreak have been excluded from incrementing minstret

shetalani, 2020-10-16:
Hi @Silabs-ArjanB, the issue is now fixed for illegal, ecall and ebreak.

closed 2020-10-16 by shetalani