openhwfoundation/cv32e40p #537: Remove  apu_master_type_o, apu_master_ready_o pins and the master from the interface name

davideschiavone, 2020-10-01:
To do after RTL freeze

MikeOpenHWGroup, 2020-10-19:
How big of an impact would this be @davideschiavone?  Currently, neither the core or UVM testbenches drive/sink the APU interface, so the impact to verification would be hours, not days.

davideschiavone, 2020-10-20:
Hi @MikeOpenHWGroup , yes but we did not want to change the core interface if not needed. Of course you drive the core verification effort, so if @Silabs-ArjanB agrees, I can remove these signals.

Silabs-ArjanB, 2020-10-22:
Hi @davideschiavone I am okay with it if also https://github.com/openhwgroup/core-v-docs/issues/48 and the discussion in https://github.com/openhwgroup/cv32e40p/commit/61ef7f29948959e94331aabb02591a98de4a80b5#commitcomment-42890979 get resolved.

davideschiavone, 2020-10-27:
A PR related to the RTL is here #558 
Whereas the one related to documentation is here: https://github.com/openhwgroup/core-v-docs/pull/233

davideschiavone, 2020-10-28:
#558 has been merged. Closing this issue

closed 2020-10-28 by davideschiavone