openhwfoundation/cv32e40p #550: Implementations does not raise illegal instruction exceptions on machine-mode access to debug registers

darbaria, 2020-10-19:
## Bug Title

According to the ISA: Machine-mode standard read-write CSRs 0x7A0–0x7BF are reserved for use by the debug system. Of these CSRs,0x7A0–0x7AF are accessible to machine mode, whereas0x7B0–0x7BF are only visible to debug mode. [Implementations should raise illegal instruction exceptions on machine-mode access to the latter set of registers.](url
[MACHINE_MODE_ACCESS_TO_DEBUG_ADDR_DOES_NOT_RAISE_IILEGAL.vcd.gz](https://github.com/openhwgroup/cv32e40p/files/5401410/MACHINE_MODE_ACCESS_TO_DEBUG_ADDR_DOES_NOT_RAISE_IILEGAL.vcd.gz)
)

Found by:
App: formalISA
Assertion failure
In the attached VCD you can see csr_addr is inside {['h7B0:'h7BF] on a valid CSR decode but the illegal insn is not raised.

Silabs-ArjanB, 2020-10-19:
Hi @darbaria the trace does not contain CSR accesses to 7B0-7BF. I thought I mentioned before that I think the assertions are probing the wrong signal (i.e. I guess the assertions are combing ID and EX stage signals.) Just look at instr_rdata_i in ID stage and see if you can find a CSR instruction there to 7B0.

Silabs-ArjanB, 2020-10-22:
Hi Ashish,

In the stated waveform there is no a single CSR instruction that uses an address in the 'h7B0:'h7BF range. The assertion must be combining signals from different pipeline stages resulting in a false positive. The `CORE.csr_addr signal is a signal that is valid in the EX stage and it can toggle also when there is no CSR instruction (which in itself is not nice but that is a different topic). To decide if an instruction is a specific CSR instruction just only use instr_rdata_i from cv32e40p_decoder.sv

 

Best regards,
Arjan

Silabs-ArjanB, 2020-10-27:
Hi @darbaria I am putting this issue as resolved because the supplied waveform does not show the suspected bug you are mentioning. Also the RTL actually has code to raise an exception for machine mode  access to specific debug registers (but your waveform does not hit that RTL code). Can you please close this issue or provide a waveform that matches the title of this issue?

darbaria, 2020-10-28:
@Silabs-ArjanB your previous point was that the CSR address was not correctly read from the pipelined stages. I consulted on this with @davideschiavone  and as per his suggestion made an amendment to the assert which again failed. I had sent an email to him on 22nd but I have not heard from him. Although my assertion does not explicitly checks for exception, I expect to see the illegal_insn go high. So far, all the feedback I had got from both of your independently was on not picking the correct csr address.

darbaria, 2020-11-03:
@MikeOpenHWGroup although this ticket was tagged resolved earlier, from my point it is only resolved now, I have updated today's email discussion here, as @davideschiavone believes there may be another related RTL issue. So, at this point I'm not certain what I had originally written was certainly an assertion issue, but I've found another way to model the assert that does establish what I was after.

@davideschiavone you may raise a new ticket, if you need my help with validating design change let me know.

Summary

Note that the suggestion on which precise signal I should be using for modelling my check was different between @Silabs-ArjanB suggestion and @davideschiavone. 

I had used Davide's suggested signal but there seem to be other dependencies that may be preventing the property to match the expected design behaviour. I don't want to use the new suggestion of tapping two fine-grained internal signals from RTL, A hint was provided by @Silabs-ArjanB so I have rewritten the assert differently using only arch-visible signals which happily proves. Meanwhile, Davide thinks there may be a power issue.

Silabs-ArjanB, 2020-11-03:
Hi @darbaria If I understand correctly you now also consider this issue resolved (if so, can you please close it?). I understand that as a result of your and Davide's investigation a power inefficiency (but not functional error as far as I understand the above) was spotted; it would be best if you or @davideschiavone files a separate issue for that.

darbaria, 2020-11-04:
As @davideschiavone believes the assertion that we have written is not testing this behaviour I would like to repoen this issue to address this. @davideschiavone please let me know what you find as incomplete about the assertion I sent in the email yesterday, and let me know what test you'd like to write. The original assertion I wrote which you provided more terms to add isn't the right one either according to @Silabs-ArjanB.

davideschiavone, 2020-11-06:
Dear @darbaria , in our last meeting, you replaced this condition

as_ISA_CSR_MACHINE_MODE_DBG_ACCESS_EXCEPTION:

      (is_a_csr_rsi ||

       is_a_csr_rs  ||

       is_a_csr_rc  ||

       is_a_csr_rci ||

       is_a_csr_rw  ||

       is_a_csr_rwi) && valid_instruction && !axiomise_illegal_instr && `CORE.csr_addr_int[9:8]==2'b11 && `CORE.csr_addr_int inside {['h7B0:'h7BF]}

        |->

      ##[0:2] (`DECODER.csr_illegal || `CORE.illegal_c_insn_id));

with this condition

`(axiomise_instr[6:0]==AXIOMISE_OPCODE_SYSTEM  && axiomise_instr[14:12] !='h0 &&

       axiomise_instr[29:28]==2'b11  && axiomise_instr[31:20] inside {['h7B0:'h7BF]} |-> ##[0:2] axiomise_csr_illegal); `

Which are not equivalent in my opinion, the first one is wrong but stronger.

As suggested in the email, to verify the first one, you have to replace 

CORE.csr_addr_int

with

ID_STAGE.alu_operand_b

and also that you are not in a shadow of a taken branch, so CONTROLLER.is_deconding_o must be 1 as well

darbaria, 2020-11-06:
Hi @davideschiavone number of points:

It is quite clear that you and @Silabs-ArjanB are not aligned on this one. I attach text from Arjan's email on 2 Oct here stating clearly that my original coding and the one you are suggesting are not correct. It is quite clear to me that the specification is pretty ambiguous on this one and that is one reason why the three of us cannot agree completely (though we do agree in parts). 

Your suggestion to me this morning doesn't match up with what you wrote in the email to me on 3 Nov. I have already written what you suggested to me this morning and it has failed and you have acknowledged this already in the mail to me on 3 Nov with your reasons (potentially hinting another RTL issue that may be preventing the modified assertion to pass).

You say is_decoding_o must be high, if you read the email I sent you on 22 Oct again, you will see I incorporated your suggestion and I had a failed assertion. That email is recorded below:

```

Hi Davide

Thanks for coming back on this. As you can see from the attached snapshot and the VCD I sent yesterday the value of both the D and the Q side of the flops at the cycle when property wakes up 17-19 are the same for alu_operand_b_ex signals. Also, Arjan separately said to me that I was decoding the signals in the wrong stages or mixing them up. I don’t quite see it that way. ex_valid_o, is_decoding_o they are all set to 1 at the time CSRRCI is decoded in 17-19. 

If you come to our sync up tomorrow we can look at this together. 

Ashish
```

So, @MikeOpenHWGroup,  from a ticket classifier point of view this is not a Component:Verif issue. It is either a DOC issue or an RTL issue that has surfaced as a consequence of this analysis (see point 3 below). 

Arjan's email to me on 2 Oct:
```
Hi Ashish,

Your assertion is not okay; you are combining signals from different pipeline stages. In the stated waveform there is no a single CSR instruction that uses an address in the 'h7B0:'h7BF range. The `CORE.csr_addr signal is a signal that is vlid in the EX stage.

Best regards,
Arjan

```
2. I do however agree with @Silabs-ArjanB that we should be using the instr_rdata_i signals to write this property (or at least one version that should not fail), I believe this aligns with how the rest of the ISA verification is done extracting instruction bits and making claims based on those. So the check I have works for now but a good question you have raised is whether it is complete? I do want to adddress this and hence this ticket is open.

3. Clearly, the two properties the one that proves and the one that doesn't are not equivalent. Your suggestion to me in your last email on this mentions including other side-band signals alu_en_o which I'm not convincesd is the righr *arch* check to write. It may be that for CVE4 due to the issue you found with clock-gating (also in the same email) it is needed. But this is not what you write today in this ticket. You have left out an important signal alu_en_o from this morning's update. Here is your email to me from 3 Nov. I also want to point out that your last email gives me the impression that there may be an issue in the RTL that's preventing your suggested property to work (as you say in the email).

Your email to me on 3 Nov:

> 
> Hello Ashish,
> 
> looking at the VCD file, I still see that you use the value from Q flip flop instead of D, but even if it was right, I see that when the CSR instruction is in the ID stage, you have a taken branch, thus the instruction is not executed.
> 
> I know you see id_valid_o equal to 1, but a further constraint is needed, which is alu_en equal to 1 (and not alu_en_ex_o, otherwise you use a signal that refers to the EX stage and not to ID stage)
> 
> https://github.com/openhwgroup/cv32e40p/blob/master/rtl/cv32e40p_id_stage.sv#L1508
> 
> The branch instruction is setting alu_operand_b_ex_o to 7B0, this guy stays as it is also after the CSR instruction, because the alu_en prevents updating registers.
> Now comes the csr_access_o signal from the DECODER, IMO, this guy should not be 1, i.e. , it should be gated by the branch or deasser_we_i in general. I ask Arjan to give his opinion as well. This signal is not modifying the reg-file so it is ok if we do not gate it. However, it consumes extra power as it switches the CSR and other muxes in the EX stage.
> 
> So in my opinion, you assertion should be 
> 
> as_ISA_CSR_MACHINE_MODE_DBG_ACCESS_EXCEPTION:
>   `ap (
>       (is_a_csr_rsi ||
>        is_a_csr_rs  || 
>        is_a_csr_rc  ||
>        is_a_csr_rci ||
>        is_a_csr_rw  ||
>        is_a_csr_rwi) && valid_instruction && !axiomise_illegal_instr && `ID_STAGE.alu_operand_b[9:8]==2'b11 &&  `ID_STAGE.alu_operand_b inside {['h7B0:'h7BF] **&& ID_STAGE.alu_en**} 
>         |-> 
>       ##[0:2] (`DECODER.csr_illegal || `CORE.illegal_c_insn_id));
> 
> **or maybe,if Arjan and I decide to fix the csr_access signal** , even better to replace ID_STAGE.alu_en with ID_STAGE.csr_access
> 
> Any feedback?