openhwfoundation/cv32e40p #552: ADDI Doesn't deliver the expected results if current decoded ADDI is in the shadow of a prior illegal instruction

darbaria, 2020-10-19:
Bug description:

ADDI Doesn't deliver the expected results if the current decoded ADDI is in the shadow of prior illegal instruction. When an extra condition to mask the detection of illegal_insn in the current cycle is added in the antecedent, we see proof that ADDI works correctly. I have seen a few other instructions behave this way (a handful) but not all instructions.
[ADDI_fails.vcd.gz](https://github.com/openhwgroup/cv32e40p/files/5401516/ADDI_fails.vcd.gz)

Found by: formalISA
Assertion failure
VCD attached.

davideschiavone, 2020-10-20:
Hello @darbaria ,

I see the core receives from the I-memory 

D3FFF117 

which is 

auipc	sp,0xd3fff

Then receives 00080008

which is considered compressed but illegal. So the core does not execute any ADDI.

The Compressed decoder tells the ID stage that we have fetched 

00010513 (uncompressed version of the illegal instruction, it has no meaning anyway)

which looks like ADDI x10, x2, 0

but in reality is the illegal 0008 instruction.

In fact the core jumps to the illegal routine. I do not see any ADDI executed.

darbaria, 2020-10-20:
@davideschiavone thank you for your explanation. It matches with mine too, but if you recall the aysmmetry discussion here it has become apparent. So for ADDI we can see a correct failure (not an RTL bug) because the assertion's antecdent doesn't capture the condition that the decoded instruction is not illegal. However, for many other cases, I don't need this side condition and the formal tools are able to prove that no instruction interleaving can break those properties. Why is it then that the design in this case is treating ADDI in a different way?

davideschiavone, 2020-10-21:
The design in this case is not treating an ADDI differently. It's an internal condition that you see but the core has not fetched an ADDI, so we cannot say it is executing it.

From outside, the core fetched

0xD3FFF117

and then 

0x00080008

The fact that the IF stage gives to the ID stage 

0xD3FFF117

and then 

0x00010513

it's an internal behaviour that is not enough to be judged. As indeed the IF stage tells to the ID stage whether the instruction coming is legal or not. Yes in the first case, No in the second one.

So the "ADDI" you see has no meaning for the ID stage and it is ignored it. You cannot say the core is treating ad "ADDI", the core is treating 0008 as illegal instruction.

darbaria, 2020-10-21:
After the 1:1 with Davide, it is agreed that for ADDI to work correctly on this core, we need to guard against the shadow illegal instructions. This asymmetry is due to power-related design optimizations that affect ADDI, SRLI, SRAL and SRLI apart from CSR instructions. Other user-instructions are not impacted by this. They work correctly without needing the guard on illegals.

closed 2020-10-21 by darbaria

davideschiavone, 2020-10-22:
Hi @darbaria , I would write like this (or at least I understood like this).
The ADDI works on this core anyway. To formally prove this, we need to guard against the shadow illegal instruction.
This is due to the fact that the compressed decoder can emit this kind of instructions (ADDI, SRLI, SRAL SRLI) together with the illegal pin, which makes them to be threated like a _don't care_

darbaria, 2020-10-22:
@davideschiavone the only point I would like to add is that it is not formal for which we need this. In fact, you would need this in a simulation TB too. This is just what you want to control and observe. If I choose to add this guard for all instructions which I bet simulation would have done, you will never see this artefact at all. When we play with coverage in formal context, we can uncover this.