openhwfoundation/cv32e40p #553: FENCE

darbaria, 2020-10-19:
Bug description: In my understanding FENCE is supposed to act as a NOP for cv32e40p. If this is so, then like a NOP instruction I would expect it to increment the PC and leave everything else unchanged. I'm able to prove that current and next-PC both remain stable on an issued FENCE. 

No traces to show but a proof exists showing the above.

Although I can prove other useful artefacts as well but this doesn't seem right. Note, I'm not talking about FENCE.I  that behaves correctly as per the specification.

If my understanding of FENCE is not correct, please advise on what is the correct specification for FENCE for this core. If my understanding is correct, then this is an RTL issue, and it should be fixed.

Silabs-ArjanB, 2020-10-19:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Hi, can somebody just simulate that instruction. I am quite sure you will see that the pc increments; maybe the issue is that the increment comes a couple of cycles after the assertion expects it.

MikeOpenHWGroup, 2020-10-19:
@darbaria, @Silabs-ArjanB, we definitately have NOPs and FENCEs running in simulation.  If the PC did not increment then the simulation would hang.   AFAIK, the FENCE and NOP instructions should also increment mcycle and minstret counters (if enabled).  By coincidence I am looking at the functional coverage for FENCE at the moment and I see that we do not enable either mcycle or minstret counters in any of our test-programs that contain a FENCE.  This is an over-sight which I shall fix (and a good demonstration of the value of functional coverage).

I'll attached waves to this issue to show that we have proper PC increments with a FENCE.

Nevertheless, I am not prepared to say that the PC is incremented properly by a FENCE under all conditions - only that it has incremented properly as we have exercised it in simulation.  So it would be useful to get a trace from formal.

darbaria, 2020-10-19:
@MikeOpenHWGroup I don't see a wave.

MikeOpenHWGroup, 2020-10-19:
@darbaria, @Silabs-ArjanB, even compressed the waves are PDB, so I will email both the waves and the test-program that illustrates that both fence and nop advance the PC.

darbaria, 2020-10-20:
I have a proof about FENCE incrementing the PC. Although what I noticed earlier is also valid the PC increments as well. Following properties are all invariants on the design.
```
as_ISA_FENCE_1 : `ap (is_a_fence && axiomise_instr_valid   |=> $stable(axiomise_c_pc));

as_ISA_FENCE_2 : `ap (is_a_fence && axiomise_instr_valid   |=> $stable(axiomise_n_pc));

as_ISA_FENCE_3 : `ap (is_a_fence && axiomise_instr_valid   |=> axiomise_n_pc==$past(axiomise_n_pc));

as_ISA_FENCE_4 : `ap (is_a_fence && axiomise_instr_valid   |=> axiomise_n_pc==$past(axiomise_c_pc)+4);
 
as_ISA_FENCE_5:  `ap (is_a_fence && axiomise_instr_valid   |-> axiomise_n_pc==axiomise_c_pc+4);

closed 2020-10-20 by darbaria