openhwfoundation/cv32e40p #554: Deadlock@ ctrl_fsm_cs stuck in DECODE in presence of conditional and unconditional branches

darbaria, 2020-10-19:
Bug description:

This is simply to track this issue. Details have been provided seperately.

Found by: formalISA
Assertion failure

ctrl_fsm_cs can get stuck in DECODE. An external interrupt can not get it out. This only happens when the CPU decodes a sequence of conditional and unconditional branches.  On masking the decode of these branch instructions, we see no failures in a 72-hour formal run. Having used these checks before verifying deadlocks in other cores, we have a strong reason to believe this can be a potential problem - the only way to get it out of the DECODE is to reset the core.

davideschiavone, 2020-10-21:
Hello @darbaria , can you send me (here or by email) the VCD file?

Silabs-ArjanB, 2020-10-22:
Hi @darbaria Thank you for the report. This seems to be a false positive (i.e. issue with the assertions itself). The provided case is simply a 'jump to itself without having first locally/globally enabled interrupts'. In that case you indeed need a reset or a debug request to get in a different ctrl_fsm_cs state, which is perfectly fine (the core is executing the 'while(1) loop' as it should.

darbaria, 2020-10-22:
@Silabs-ArjanB in the VCD I provided, both debug and interrupts are enabled. I can see that interrupt can wake up the core if it was executing OPCODE_SYSTEM. So, they're not blocked in formal. The reason the core doesn't exit the conditionals is because there is nothing in the design to pick up the gloabl interrupt to cause the loop to be broken. I would not raise this as an issue if irq_i and debug_req_i were tied off. Lastly, I don't see this behaviour in other cores from the same family - we saw that when debug was enabled and there were issues. But in general, using reset to bring the core out of lockup is not something I'd agree with.

Silabs-ArjanB, 2020-10-22:
Hi @darbaria In the provided VCD interrupts are generated at irq_i, but that does not mean that interrupts are locally/globally enabled inside the core. They are not and therefore it does not matter that irq_i is toggling. Other cores might implement non-maskable interrupt functionality which would behave differently. In the provided waveform debug_req_i is always low. If you can provide a waveform with locally/globally enabled interrupts and/or a debug_req_i that is high and you show that ctrl_fsm_cs remains in DECODE, then there would be an issue (but we need a different waveform for that).

darbaria, 2020-10-22:
Hi @Silabs-ArjanB which registers would you need to set to see the irq locally enabled? If you give me an example of what you'd expect to see from a register state point of view to see these unmasked, I can look at it.  As for debug, it is not that they're disabled, but formal tool hasn't been able to find a way to power it to 1 to get the core to come out of DECODE. I have a requirement that debug must toggle.

Silabs-ArjanB, 2020-10-23:
Hi @darbaria local enabling of interrupts is done via the 'mip' CSR. Global enabling of interrupts is done via the 'mie' bitfield in the 'mstatus' CSR. With respect to debug: It is hard to believe the core will not come out of DECODE when it is not in debug mode already (!debug_mode_q) and debug_req_i is set to 1 for at least one cycle. I can only look at that further if there is actually a trace showing that the core stays in DECODE with !debug_mode_q and debug_req_i being raised. Once in debug mode software is supposed to eventually issue a dret to come ouf of debug mode again. If software/debugger fails to do so, e.g. by executing a jump to itself while already in debug mode, then I would expect the core stays in DECODE as well (which would be fine).

Silabs-ArjanB, 2020-10-27:
Putting this issue as resolved as interrupts were clearly not locally/globally enabled in the provided waveform and no waveform has been provided showing being stuck in DECODE with debug_req_i = 1.

darbaria, 2020-10-30:
@Silabs-ArjanB I reviewed the settings for irq and debug. I also looked at local/global enabling of irqs. Interrupts seem to be firing up. The lockup is not because of them. You're right in stating that if the machine is in debug mode and there is a sequence of infinite JAL or the family of branch instructions, the ctrl_fsm_cs will be stuck in DECODE. I temporarily assumed that we cannot have a sequence of such instructions (no conditional or conditional branches) when the core is in debug mode and the assertion has not fired so far. One way of breaking the deadlock is to get the core out of debug mode via the dret/mret. I was curious to see why we cannot get the core out of debug mode in this case, so I wrote further tests to check the behaviour of debug_mode_q flop. 

It appears that debug_mode_q can go 0 once it has become 1 sometimes (cover pass) , but not always (assert fail). For example one trace I have is when the debug_mode_q is 1 and you have executed an infinite trace of ecall then the core's debug_mode_q can never become 0 which I believe is expected. Can you confirm if this is legal behaviour? If yes, we can ignore this, but if not I'm happy to file a new ticket.

One other question that remains unanswered:

1. Why is that an infinite sequence of all other instructions do not cause a lockup in DECODE state in presence of irq and debug?

Silabs-ArjanB, 2020-10-30:
Hi @darbaria ecall is specifically listed as an instruction with undefined behavior when executed in debug mode (and I am not surprised if you do an infinite sequence of them that you stay in debug mode, but as said the behavior is specifically stated as 'undefined' in the official debug spec).

With respect to you other question: The core simply leaves the DECODE state to handle specific instructions (e.g. an ebreak instruction or an WFI instruction). If you would only ever feed ADD instructions into the core after having properly enabled interrupts, then the FSM would also always stay in DECODE always (assuming that the interrupt service routine also only consists of ADD instructions and no debug request comes in) (and this contradicts what you imply in your question I think). So instead of focusing on what you call lockup I think it would make more sense to ask the question: Will the core always eventually handle an incoming interrupt when that interrupt is locally and globally enabled, or, will the core always eventually enter/exit debug mode when so requested.

darbaria, 2020-10-30:
@Silabs-ArjanB the core is not stuck in DECODE state when executing other instructions (e.g., ADD or others).  The whole point of my questions is why?  For last many week this is what I have been saying. Checking the behaviour of core when triggering interrupts and debug is a separate verification question, but checking that it doesn't lock up is another. For this thread, I'm only looking at answering the latter.

Silabs-ArjanB, 2020-10-30:
Simple answer: If you execute certain instructions (e.g. WFI) you will exit DECODE.

darbaria, 2020-10-30:
That is now what I see. When conditionals are disabled and other instructions get injected I don't see a failure for the assertion. Anyways, as this is not an issue for you, let's leave it at that. For me, this remains an open question!

closed 2020-10-30 by darbaria