openhwfoundation/cv32e40p #562: ebreak instruction executed during mulh in debug mode creates PC confusion

tzwaenn, 2020-10-30:
### Component

**Component:RTL**: For issues in the RTL (e.g. for files in the rtl directory)

### Steps to Reproduce
You see the FSM acting on the ebreak while the mulh is running - and the PC after the mulh is the one of the mulh (in my earlier email, I though our check was failing because the PC gets incremented by 8 - on a closer look, it seems not be incremented at all which indeed would be an ISA violation).
[image]
[RV_chk.ops.RV32M.vcd.zip](https://github.com/openhwgroup/cv32e40p/files/5465403/RV_chk.ops.RV32M.vcd.zip)

Silabs-ArjanB, 2020-10-30:
The PC after the ebreak execution is (also) equal to the dm_halt_addr_i input, which happens to be equal to the PC of the mulh instruction. Paul, can you please judge if this is an actual RTL issue or not (for now I classified it as such, but I actually do not see an ISA violation yet)?

tzwaenn, 2020-10-30:
Hi Arjan,

Good point – I was not looking for “external” reasons when my next PC check was off by 4 – but as mulh is the first instruction in debug mode, it was fetched from the entry address, and an additional ebreak in debug mode is supposed to go back to that address. So I guess executing this instruction while the mulh is ongoing is not doing any harm after all although it seemed strange that only this very particular combination of ebreak and debug mode in conjunction with mulh was flagged by the assertion...

Silabs-ArjanB, 2020-10-30:
Hi @tzwaenn If you now think that this is actually an error or oversight in the used assertion, please let us know and I will change the issue type to 'component:verif' instead of 'component:rtl' to indicate that this is not an actual RTL bug (and we will save Paul some time from diving into this)

tzwaenn, 2020-10-30:
Right, I will exclude this special case for now until we actually model in our verification environment that an ebreak gets executed while a mulh is still running.

Silabs-ArjanB, 2020-11-02:
Hi @tzwaenn I change the type of this issue to indicate that it is an assertion issue as opposed to an RTL issue. Please let us know once it has been resolved.

MikeOpenHWGroup, 2020-11-11:
Hi @tzwaenn, can you have a look at this one?

tzwaenn, 2020-11-12:
This is worked around in our environment. A clean verification of the ebreak here overtahing the mulh is a separate matter.

closed 2020-11-12 by tzwaenn