openhwfoundation/cv32e40p #563: CV32E40P does not signal  signal 'halted', 'running', 'havereset' to the Debug Module

Silabs-ArjanB, 2020-11-02:
The interface between the PULP Debug Module (https://github.com/pulp-platform/riscv-dbg) and the CV32E40P is not as it should be. In short the CV32E40P does not have a proper way to signal 'halted', 'running', 'havereset' to the Debug Module as is required from the Debug spec:

[image]

In the current implementation the Debug Module _itself_ has a notion of whether the RISC-V core should be considered 'halted', 'running', or 'havereset'. Unfortunately the RISC-V state as thought of by the Debug Module can easily become out of sync with the actual CV32E40P state as is reported in https://github.com/pulp-platform/riscv-dbg/issues/65. Only the CV32E40P will know whether it has actually been reset; the Debug Module cannot assume that this happened for example because it caused an ndmreset.  

The proper way to address this issue would be to make the CV32E40P itself responsible for signaling 'halted', 'running', 'havereset' on its interface (as written in the RISC-V debug spec). These signals are expected to be easily available (or constructable) from within the CV32E40P, e.g. 'halted' is equal to the existing debug_mode_q flop.

MikeOpenHWGroup, 2020-11-02:
Excellent catch @Silabs-ArjanB.  While this will have a direct impact on the cv32e40p project, it does seem like we should address it now, rather than later.   As far as I am aware, the PULP Debug Module does not accept  'halted', 'running', or 'havereset' inputs from the core, so this may to be addressed as well.

Silabs-ArjanB, 2020-11-02:
@MikeOpenHWGroup I think we should at least add the required pins already now and then in parallel open the discussion with the PULP team. If later (after RTL freeze) the interface definition changes slightly we would at least already have the pins. The current debug module can still be used even if it does not use the pins we intend to add (it would simply have the already reported bug).

Silabs-ArjanB, 2020-11-02:
Resolved with https://github.com/openhwgroup/cv32e40p/pull/564. @silabs-PaulZ can you please review the PR?

bluewww, 2020-11-02:
The debug module interface was afaik chosen like it currently is because it mirrors what the rocket debug module does. I believe we should first try to understand what rocket does and how they don't need these signals

silabs-PaulZ, 2020-11-02:
Hi @bluewww ,

The core can get a reset independent of the ndmreset from the debug module. In this case, the PULP debug module will have an incorrect havereset status. The debug module state may also be out of sync with the core sate; The debug module may be in HALTED state while the core is UNAVAILABLE due to reset or came out of reset into RUNNING state.

These new signals allow the core to be compliant with the spec and work with other debug modules. The PULP debug module will continue to work as-is without using these signals..though there seems to be a bug with PULP debug module if an external reset is applied to the core and not the debug module.

Silabs-ArjanB, 2020-11-02:
Hi @bluewww , here are some parts of the RISC-V Debug spec that show that the Debug Module cannot really predict/know the core's debug status as is currently the assumed in the PULP Debug Module:

[image]

[image]

bluewww, 2020-11-02:
@Silabs-ArjanB Thanks. The second to last sentence looks quite convincing that cores need to report their state directly instead of letting the debug module infer it.

bluewww, 2020-11-02:
@silabs-PaulZ 
Yeah I guess we didn't encounter the bug yet because only the debug module can reset the cores independently from the system reset.

PR #564 (2020-11-03): Added debug status pins; fix for #563
Fixes #563 

- Example waveforms showing the behavior of the added signals (to be added in documentation for CV32E40P):
- debug_halted_o has been shown to be equivalent to debug_mode_q (assertion is present in RTL)

[image]

[image]
closed 2020-11-03 by Silabs-ArjanB