openhwfoundation/cv32e40p #565: CSR MRW registers raise CSR illegal on a read

darbaria, 2020-11-02:
@davideschiavone Following on from my email earlier the issue reported is indeed a bug. 

RTL: Slightly older from late Sep.
Found by: formalISA and JasperGold

The decoder flags csr_illegal when accessing several CSR RW registers e.g., from the family MRW (that are implemented on CVE4) . This is due to a design issue at line 2623 in the decoder module. The trace I sent was for MISA but other registers are equally affected. 

On locally patching the RTL, I'm able to prove the about 9 asserts that state that we must not flag csr_illegal when accessing these registers. 

The patch seems to be fix the issue and doesn't break all the other asserts on illegals and exceptions. 

[MISA_ILLEGAL.vcd.gz](https://github.com/openhwgroup/cv32e40p/files/5478355/MISA_ILLEGAL.vcd.gz)

MikeOpenHWGroup, 2020-11-02:
Thanks for this bug report @darbaria.   I hate to be a pain, but the reference to the RTL is not terribly useful:

> RTL: Slightly older from late Sep.

Please provide the git hash for your clone.  To get that, set your cwd to somewhere (anywhere) in the cloned repo and type:
```
git branch -v
```
The seven digit number is what we need.

darbaria, 2020-11-02:
@MikeOpenHWGroup it is d0ad787

darbaria, 2020-11-02:
@MikeOpenHWGroup I didn't clone it, I downloaded the master as a zip. It is from 25 Sep. However, the latest RTL I see also has the same code at line 2356.

darbaria, 2020-11-03:
Just capturing the key points discussed in the meeting. The specification mentioned on Page 131 of the unprivileged spec does not talk about the case when 13:12==2'b00 so the designer has decided to flag this case by using the default branch. There is nothing in sepc to suggest we should be flagging this as illegal. If we are flagging this as illegal, then we shouldn't be setting the is_decoding_o and id_valid_o signals high in the cycle where a CSR read is requested, as this means that a valid csr_op is occuring. As both conditions are occuring at the same time, the core can set the illegal pin high ignoring csr_op and causing updates to mcause and mepc.

So either we relax the case of not flagging the csr_illegal or think about fixing the is_decoding_o and id_valid signals so we don't detect a valid csr_op (is_decoding_o high) for an illegal csr.

Silabs-ArjanB, 2020-11-03:
@davideschiavone With respect to above remark "If we are flagging this as illegal, then we shouldn't be setting the is_decoding_o and id_valid_o signals high in the cycle where a CSR read is requested, as this means that a valid csr_op is occuring " do you see a functional RTL issue here? For non-CSR instructions (I believe) we do not suppress is_decoding or id_valid when the instruction is invalid. The only thing that needs to be guaranteed is that there are no side effect as for example register file writes or CSR writes for any invalid instruction. Are any unwanted side effects actually happening in the scenario of CSR instructions that have bits 13:12 == 2'b00 or is it just that csr_op remains set which might look confusing but would still be harmless?

davideschiavone, 2020-11-04:
Summarizing a bit what is going on by email, meeting and GitHub issues:

** CSR machine mode debug check **

This issue is not complete. The formal check has been rewritten but not completly and it is different than the original question.

** Signaling illegal instructions if CSR operation has as funct3 field bit 13:12 = 2'b00 **

According to the designers, this is ok to be treaten like an illegal. The design works as intended. However, if the RISC-V ISA is not clear about that, another ticket should be open about the interpretation of the ISA by the designers,  rather than an RTL bug - as RTL behaves like the designer intended to. 

Closing the issue right now, but of course feel free to re-open and/or reply here

closed 2020-11-04 by davideschiavone

darbaria, 2020-11-04:
@davideschiavone 

As regards #565

Why has this issue been closed?

Why is the RTL tagging a valid instruction as illegal? If the condition for classifying the CSR instruction as valid is not that is_decoding_o has to be high then what is this condition? If the instruction is correctly being decoded as illegal then I need to know why is_decoding_o is high and if it is meant to be high then where is this documented?

As regards the other issue I will respond in that thread.

Silabs-ArjanB, 2020-11-05:
The stated instruction is not defined and as such it is allowed (but not mandated) by the RISC-V spec to be treated as an illegal isntruction:

[image]

darbaria, 2020-11-05:
Arjan thanks for adding some insight on this. 

Why is the design tagging these as valid in the same way as the rest of the decoder i.e., is_decoding_o and id_valid both high? 

Honestly, I could not get much sleep last night thinking about this. Why can’t the design tag this case as invalid? If there is a good reason for this I would like to know. 

BTW I can confirm that the writes in this special case are not doing any nasty stuff so you’re okay. So this also means that if you don’t do anything for the default branch you will achieve the same thing as you do now.  My view is it saves you extra operations on exception updates etc.

Silabs-ArjanB, 2020-11-05:
The id_valid signal just means there is valid info in a stage available for the next pipe stage; it simply does not imply anything about whether an instruction is invalid or not. It is not meant to be the inverse of the invalid signal