openhwfoundation/cv32e40p #571: Single stepping through hwloop

bluewww, 2020-11-07:
Single stepping through hardware loops doesn't work.

### Steps to Reproduce
1. git hash 6c858bca4559da3b191007ce2f1debeadbeb635e (master)

Unfortunately, I don't have a minimized reproducible example at hand, but it was reported to me that single stepping via openocd through hwloops doesn't work. Also, back then when I implemented the states for riscv debug module support, I didn't invest any resources to add the special handling probably required for hwloops.

MikeOpenHWGroup, 2024-06-04:
Hello @bluewww, it has been a very long time and I know this is a long shot, but are you in a position to replicate this issue?  At this point the HWLOOP feature has been extensively verified and there are no known RTL issues related to this feature.

pascalgouedo, 2024-06-05:
Hi @bluewww 
As seen during Hardware loop verification, special code was added in debug program to manage the end of the loop. But it was required only for ebreak/ecall and no specific code was needed for Single-Step.
You can have a look to 6.2 section of [latest CV32E40Pv2 User Manual](https://docs.openhwgroup.org/projects/cv32e40p-user-manual/en/latest/corev_hw_loop.html#hardware-loops-impact-on-application-exception-handlers-and-debug-program).
I can't be sure but I don't think anything special is required in debug modules either.

I will waive this issue but feel free to re-open it if you are able to reproduce this problem with CV32E40P v1.8.0 release.

MikeOpenHWGroup, 2024-06-05:
Thanks @bluewww!

pascalgouedo, 2024-06-05:
Hi @MikeOpenHWGroup 
@bluewww didn't answer, I added additional information and decided to waive.
Pascal.

closed 2024-06-06 by MikeOpenHWGroup

bluewww, 2024-06-21:
Sorry I didn't see you guys. I haven't restested for the bug again but @pascalgouedo's explanation seems sensible to me.