openhwfoundation/cv32e40p #572: Possible lint issue in latest RTL (6c858bc)

MikeOpenHWGroup, 2020-11-10:
With the updated update to the RTL (hash 6c858bc), Metrics **dsim 20200720.14.0** complains about the following three lint issues:
```
=W:[LatchInferred]:
    The following variables in always_comb blocks are not written    
    along all paths through the block.  Latches will be inferred:    

    /data/mike/GitHubRepos/MikeOpenHWGroup/core-v-verif/master/core-v-cores/cv32e40p/rtl/cv32e40p_register_file_ff.sv:109:3    we_a_dec                 

    /data/mike/GitHubRepos/MikeOpenHWGroup/core-v-verif/master/core-v-cores/cv32e40p/rtl/cv32e40p_register_file_ff.sv:119:3    we_b_dec                 

    /data/mike/GitHubRepos/MikeOpenHWGroup/core-v-verif/master/core-v-cores/cv32e40p/rtl/cv32e40p_cs_registers.sv:1439:3       mhpmcounter_increment    
```
The first two have been discussed previously.  I cannot convince myself that the third is valid, but thought I should bring it to your attention in any case.

Silabs-ArjanB, 2020-11-10:
That looks like a dsim issue to me. The stated signals will not be latches.

MikeOpenHWGroup, 2020-11-10:
I thought so as well @Silabs-ArjanB, but felt it was worth a quick review.  Thanks.

closed 2020-11-10 by MikeOpenHWGroup