openhwfoundation/cv32e40p #583: HWLoop minimum size 3 illegal exception

davideschiavone, 2020-11-18:
As the HWloop body must be at least 3 instructions, the core should raise an illegal exception when this is not respected

In addition, as the lp.starti and lp.endi instruction use the immediate by aligned implicitly (<<1) as specified here

https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/instruction_set_extensions.html#operations

we should change their definition by forcing word alignment (<<2) as it is a constraint of the HWloop

https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/pulp_hw_loop.html

Pinging @jeremybennett , @jm4rtin  and @eflamand

davideschiavone, 2020-12-04:
Before proceeding and updating the documentation , we want to try the RTL (by means of hexadecimal instructions) to avoid that the SW team modifies the custom HWLoop relocation before RTL gets tested as agreed with @jm4rtin and @jeremybennett

davideschiavone, 2022-12-23:
@pascalgouedo - we agreed that the Core won't trap to any exception - shall I close it?

pascalgouedo, 2023-01-03:
Hi @davideschiavone 
Yes both updates (no trap and shift by 2) have been included in v1.2.0 User Manual.

closed 2023-01-03 by davideschiavone