openhwfoundation/cv32e40p #588: Hello_world simulation stuck at time 0 using Vivado Simulator

jackieke724, 2020-11-26:
Hello,

For some "`include" problems with Modelsim Starter, I could not get your flow working. Therefore, I decided to deploy a test project in Vivado. 

I am trying to have the simple hello world simulation (equivalent to make custom-vsim-run) running with XSim, but the the simulation stucks at 0. I hope you might share some insights about my issue.

Things I have done:
1. included all the rtl and tb files except for the fpnew modules, latch_regfile, 
2. compiled hello_world.c to hex (by make hello_world.hex) and included the hex in the project
3. Vivado simulator successfully compiled and started the simulation

But then the simulation hangs at time 0.

I know this is an unfair question for you because you might not know Vivado. However, I am guessing this could be related to the files I imported to my project. Maybe I have included the wrong file or some module does not work in the Vivado Simulator. I would really appreciate any insights from you!

I have attached the simulation log, with additional output of executed process and verilog statements using "xsim -tp -tl". The last executed process is at the aligner module.  [VivadoXsim.log](https://github.com/openhwgroup/cv32e40p/files/5603172/VivadoXsim.log)

Thanks a lot!

Silabs-ArjanB, 2020-11-26:
Hi @jackieke724 Sounds a bit like a delta cycle issue (caused by somehow combinatorial blocks triggering each other and the simulator not being able to figure out that how to converge); maybe your tool has a feature to debug such issues? I don't know anything about Vivado myself, so I indeed cannot really comment on that or recreate the issue. Maybe an alternative starting point for running simulations is the https://github.com/openhwgroup/core-v-verif as a larger variety of simulators are supported in that repos.

davideschiavone, 2020-11-26:
Hello, adding to what @Silabs-ArjanB said, maybe not all the features are supported in Vivado (as for example the random interrupt generator or memory stalls etc). Try to remove those from the testbench as well.
I changed the label from BUG to QUESTION as here no BUG is reported (yet)

jackieke724, 2020-11-27:
Hello, thank you all for the inputs. I have tried to exclude the random_interrupt_generate module and the memory_stall modules from my project, using the "`define VERILATOR"  flag. The simulator still hangs at time 0. Are there any other modules/features I can exclude to simplify the testbench?

jackieke724, 2020-11-29:
I just want to leave my solution here in case someone else needs it. After my fix, I am able to simulate both the hello_world and interrupt tests. 

The problematic line for me is line 252 `automatic string error_str;` in "mem_ram.sv". Even if I comment out the rest of the code in that begin-end block (i.e. line 254-286), and just leave those instantiations for `automatic` variables there, the simulator will still stuck at time 0. I do not know why this line is not working.

At the end, I just commented out 

- `automatic string error_str;`
- `errno = $ferror(sig_fd, error_str); $error(error_str);` which needs the `error_str` variable.

Btw, I am using Vivado 2020.1.1.

closed 2020-11-29 by jackieke724

Yym996, 2021-11-25:
Hello, @jake-ke 

I'm also trying to run the simulation of hello world, but there is something wrong with it. 

I added the rtl and tb files in the project as you did. And also added the hex file only for simulation.

But it shows warning:
> [Memdata 28-176] There are no bmm files or elf files. Therefore Vivado could not produce any .mem files. Check the design for the existence of processors and associated elf files. Here are the info from TCL Console:
[image]

I started work with vivado not long ago, have no idea what could be the reason for that. I would appreciate any suggestions from you.

Thanks in advance!