openhwfoundation/cv32e40p #590: Design checklist items for Functional RTL Freeze

MikeOpenHWGroup, 2020-11-30:
@davideschiavone, as part of the sign-off of Functional RTL Freeze for CV32E40P, please complete the following items in the [Design Checklist](https://github.com/MikeOpenHWGroup/core-v-docs/blob/master/program/milestones/CV32E40P/RTL_Freeze_marchid4_mimpid0/DesignChecklist.md):

- [ ] Create a tag of the RTL
- [ ] Sign-off all items in "coding rules"
- [ ] Sign-off all items in "design rules"
- [ ] Sign-off remaining "open source" items (we will need to engage with Eclipse for this)

This issue can be closed when all three items in the Checklist are signed-off by you or your delegate.

Silabs-ArjanB, 2020-12-04:
Please also add the following information in the checklist:

- [ ] Number of most recent github issue in the cv32e40p repos (either open or closed, so that we will later know which issues were known at time of RTL freeze).
- [ ] Complete list of github issue references/numbers of waived issues (independent of how/why the issue was waived)

davideschiavone, 2020-12-08:
Hi @Silabs-ArjanB , I update the documentation to add those 2, maybe it would be good to have a Design, Verif and Doc separate files to report all the issues as here: https://github.com/davideschiavone/core-v-docs/blob/task261/program/milestones/CV32E40P/RTL_Freeze_v1.0.0/Design_openissues.md

MikeOpenHWGroup, 2020-12-09:
@davideschiavone, After discussions with @DBees, we are recommending that the last check-box on this issue (Sign-off remaining "open source" items) be waived.  These review with Eclipse is likely to spill into the New Year, and I do not recommend gating Functional RTL Freeze on it.

Silabs-ArjanB, 2020-12-10:
Agreed with Davide to not waiting for 'Sign-off remaining "open source" items (we will need to engage with Eclipse for this)'. Closing this issue.

closed 2020-12-10 by Silabs-ArjanB

DBees, 2020-12-10:
Agree but note that an additional IP checklist was created to replace these rows which were removed from Design checklist. The major element of the IP checklist is the completion of IP review, which follows the Eclipse Contribution Questionnaire process.