openhwfoundation/cv32e40p #591: Documentation checklist items for Functional RTL Freeze

MikeOpenHWGroup, 2020-11-30:
@davideschiavone, as part of the sign-off of Functional RTL Freeze for CV32E40P, please complete the following items in the [Documentation Checklist](https://github.com/MikeOpenHWGroup/core-v-docs/blob/master/program/milestones/CV32E40P/RTL_Freeze_marchid4_mimpid0/DocumentationChecklist.md):

- [ ] Tagged version of source available in core-v-docs.
- [ ] Update released version the User Manual in RTD attributed to core-v-docs tag.
- [ ] List all waived issues attributed to tagged release, if any.

This issue can be closed when all three items in the Checklist are signed-off by you or your delegate.

Silabs-ArjanB, 2020-12-04:
Please also add the following information in the checklist:

- [ ] Number of most recent github issue in the core-v-docs repos (either open or closed, so that we will later know which issues were known at time of RTL freeze).

Silabs-ArjanB, 2020-12-10:
Discussed this with Davide and we decided to right now tag the core-v-docs with the cv32e40p_v1.0.0 tag. Note that the tag only applies to the User Manual as the checklist documents are not complete yet (and will be updated by later git commits). If we don't do this we have a chicken-and-egg problem. In the future the User Manual should move into the cv32e40p repos so that it can become tagged together with the RTL.

Also, we decided to ignore the 'Update released version the User Manual in RTD attributed to core-v-docs tag' checklist item as only the documentation source files are relevant.

closed 2020-12-10 by Silabs-ArjanB