openhwfoundation/cv32e40p #592: Simulation Verification checklist items for Functional RTL Freeze

MikeOpenHWGroup, 2020-11-30:
@strichmo , as part of the sign-off of Functional RTL Freeze for CV32E40P, please complete the following items in the [Simulation Verification Checklist](https://github.com/MikeOpenHWGroup/core-v-docs/blob/master/program/milestones/CV32E40P/RTL_Freeze_marchid4_mimpid0/SimulationVerificationChecklist.md):

- [ ] Block Code Coverage
- [ ] Expression Code Coverage
- [ ] Covergroup Coverage
- [ ] Cover-property Coverage
- [ ] Regression report on the RTL Freeze tag of the cv32e40p RTL

This issue can be closed when all three items in the Checklist are signed-off by you or your delegate.

Silabs-ArjanB, 2020-12-04:
Please also add the following information in the checklist:

- [ ] Number of most recent github issue in the core-v-verif repos (either open or closed, so that we will later know which issues were known at time of RTL freeze).
- [ ] Complete list of github issue references/numbers of waived issues (independent of how/why the issue was waived)

strichmo, 2020-12-08:
All items are published.  Closing the issue.

Silabs-ArjanB, 2020-12-10:
Unchecked all checkboxes and reopened this issue as sign off was not done on the correct version of the cv32e40p repos. The RTL freeze cv32e40p repos has git hash 120ac3ee79ef56a57fe07dd8701cd4ee94458fd5 (and verification seems to have been performed 7 commits before that)

MikeOpenHWGroup, 2020-12-16:
The core-v-verif repo now has a tag named `cv32e40p_v1.0.0` to match the equivalent tag for this repo.  This tag of core-v-verif will clone the cv32e40p_v1.0.0 tag of the RTL by default.  The coverage reports and waivers are all based on runs from that tag.

closed 2021-01-04 by Silabs-ArjanB