openhwfoundation/cv32e40p #593: Simulation Verification checklist items for Functional RTL Freeze

MikeOpenHWGroup, 2020-12-01:
@MikeOpenHWGroup  , as part of the sign-off of Functional RTL Freeze for CV32E40P, please complete the following items in the [Simulation Verification Checklist](https://github.com/MikeOpenHWGroup/core-v-docs/blob/master/program/milestones/CV32E40P/RTL_Freeze_marchid4_mimpid0/SimulationVerificationChecklist.md):

- [ ] Verification Planning: Testbench Cross-reference
- [ ] Ensure all cover-properties attributed to a Vplan item
- [ ] Ensure all cover-groups attributed to a Vplan item
- [ ] All RISC-V Compliance tests run and passing
- [ ] All other tests are running and passing in regression

This issue can be closed when all items in the Checklist are signed-off by you or your delegate.

Silabs-ArjanB, 2020-12-04:
Please also add the following information in the checklist:

- [ ] Number of most recent github issue in the core-v-verif repos (either open or closed, so that we will later know which issues were known at time of RTL freeze).
- [ ] Complete list of github issue references/numbers of waived issues (independent of how/why the issue was waived)

strichmo, 2020-12-08:
I have completed the interrupt vplan annotation and checked off the cross-reference as a result.

MikeOpenHWGroup, 2020-12-09:
To @Silabs-ArjanB's request:
The most recent github issue in the core-v-verif repo is [#425](https://github.com/openhwgroup/core-v-verif/issues/425)

The list of waivers (all captured as core-v-verif GitHub issues is:
 - [#403](https://github.com/openhwgroup/core-v-verif/issues/403)
 - [#423](https://github.com/openhwgroup/core-v-verif/issues/423)
 - [#424](https://github.com/openhwgroup/core-v-verif/issues/424)
 - [#425](https://github.com/openhwgroup/core-v-verif/issues/425)

Silabs-ArjanB, 2020-12-09:
Hi @MikeOpenHWGroup Can you please document that similar to how Davide did this? See https://github.com/openhwgroup/core-v-docs/blob/master/program/milestones/CV32E40P/RTL_Freeze_v1.0.0/Design_openissues.md

MikeOpenHWGroup, 2020-12-09:
Table added.   See https://github.com/openhwgroup/core-v-docs/pull/306.

Silabs-ArjanB, 2020-12-10:
Hi @MikeOpenHWGroup 

I reviewed this and I undid the checkboxes related to 'All RISC-V Compliance tests run and passing' and 'All other tests are running and passing in regression' as they were not run on the correct cv32e40p repos. The hash to be used is 120ac3ee79ef56a57fe07dd8701cd4ee94458fd5

MikeOpenHWGroup, 2020-12-11:
I've created a [core-v-verif branch](https://github.com/MikeOpenHWGroup/core-v-verif/tree/point_to_tagged_cv32e40p) that clones the RTL from the cv32e40p_1.0.0  tag.  Regression is clean so I have re-checked the checkboxes.

The last checkbox (Tag core-v-verif repos with cv32e40p_v1.0.0 tag) will be completed when the above branch is [merged](https://github.com/openhwgroup/core-v-verif/pull/435) in and regression has been re-run from there.

MikeOpenHWGroup, 2020-12-16:
corev-v-verif has been tagged as above.

closed 2021-01-04 by Silabs-ArjanB