openhwfoundation/cv32e40p #594: Formal Verification checklist items for Functional RTL Freeze

MikeOpenHWGroup, 2020-12-01:
@tzwaenn, as part of the sign-off of Functional RTL Freeze for CV32E40P, please update all items in the [Formal Verification Checklist](https://github.com/MikeOpenHWGroup/core-v-docs/blob/master/program/milestones/CV32E40P/RTL_Freeze_marchid4_mimpid0/FormalVerificationChecklist.md).

This issue can be closed when all items in the Checklist are signed-off by you or your delegate.

Silabs-ArjanB, 2020-12-10:
Hi @MikeOpenHWGroup , @tzwaenn , This item cannot be closed yet for three reasons. 

- I do not have a confirmation that the formal verification was actually ran on cv32e40p version 120ac3ee79ef56a57fe07dd8701cd4ee94458fd5. 
- The FormalVerificationChecklist.md checklist is not filled out completely; osmt sign off dates are missing.
- The FormalVerificationChecklist.md checklist states in the Exceptions/Waivers/Comments column that they are 'captured as cv32e40p GitHub issues'. The design team is not aware about any such captured issues. Please either remove such comments or list the related github numbers

tzwaenn, 2020-12-15:
* version 120ac3e is being run right now
* due to some resource conflicts with our own development, we had to cancel our runs and we got partial results so far (no fails, some hold bounded, some hold)
* excetions/waivers column adapted by Nicolae
* will add more details on results in the report when they appear for the new run

MikeOpenHWGroup, 2020-12-16:
Hi @tzwaenn, any update today?

tzwaenn, 2020-12-16:
Hi Mike,

the run I mentioned yesterday had some unexpected bounded results - 
-I- Prv - Intermediate: RV_chk.ops.RV32I.Arith_a holds bounded up to 24 cycles from reset
-I- Prv - Intermediate: RV_chk.ops.RV32I.CSR_a holds bounded up to 26 cycles from reset
-I- Prv - Intermediate: RV_chk.ops.RV32I.Mem_a holds bounded up to 25 cycles from reset
-I- Prv - Intermediate: RV_chk.ops.RVC.Mem_MA_a holds bounded up to 26 cycles from reset
-I- Prv - Intermediate: RV_chk.ops.RVC.Jump_a holds bounded up to 26 cycles from reset
-I- Prv - Intermediate: RV_chk.ops.xcpt_fetch_dec_a holds bounded up to 30 cycles from reset
-I- Prv - Intermediate: RV_chk.ops.RV32I.Branch_taken_a holds bounded up to 25 cycles from reset
We found yet another setup issue and now things look better:
-R- Property 'RV_chk.ops.RV32I.WFI_flush_a' holds (checked in 10 min 03 sec, 7660 MB used)
-R- Property 'RV_chk.ops.RV32I.FENCE_I_a' holds (checked in 28 min 01 sec, 8096 MB used)
-R- Property 'RV_chk.ops.bubble_a' holds (checked in 04 min 05 sec, 6488 MB used)
-R- Property 'RV_chk.ops.reset_a' holds (checked in 4.00 sec, 5101 MB used)
-R- Property 'RV_chk.ops.RV32I.CallBreak_a' holds (checked in 35 min 01 sec, 8694 MB used)
-R- Property 'RV_chk.ops.RV32I.EBREAK_BP_DM_a' holds (checked in 37 min 08 sec, 8197 MB used)
-R- Property 'RV_chk.ops.RV32I.EBREAK_DBG_a' holds (checked in 39 min 00 sec, 8504 MB used)
-R- Property 'RV_chk.ops.RV32I.RET_a' holds (checked in 40 min 02 sec, 8880 MB used)
-R- Property 'RV_chk.ops.xcpt_DBG_BP_a' holds (checked in 10 min 16 sec, 8309 MB used)
-R- Property 'RV_chk.ops.RVC.Branch_taken_a' holds (checked in 55 min 57 sec, 7980 MB used)
-R- Property 'RV_chk.ops.RV32I.Branch_taken_a' holds (checked in 1 h 00 min 18 sec, 8214 MB used)
-R- Property 'RV_chk.ops.xcpt_fetch_dec_a' holds (checked in 37 min 39 sec, 9496 MB used)
so a few of the non-converging ones were actuallz pretty fast.
The remaining results may take another iteration.
Sorry for the delay.

MikeOpenHWGroup, 2020-12-16:
Thanks for the update @tzwaenn.  I would rather do it "right" than do it "fast", so I appreciate you taking the extra time for that.

tzwaenn, 2020-12-17:
We now have all exceptions, CSRs, Arithemtic instructions, branches, memory with single access and memory with double access through (RV32I + compressed where applicable). For some reason, jump is still at hold_bounded which is very unexpected (memory instructions with 2 memory accesses were the most expensive in previous runs).
Partial tests for M extension started again.
We will investigate the setup again for jumps...

tzwaenn, 2021-01-05:
As updated in the formal regression run results for end of last year, we had full proofs for RV32I, RVC, exceptions, Zicsr, Zifencei apart from the jump instructions. The RV32I jumps got a bounded proof with 30 cycles from reset, the RVC jumps got 38 cycles from reset in that run which was performed on the requested github version of the source code. Disregarding the PC update, all jumps are fully proven (so register file update, CSR update, lack of data memory access, and ending in a clean state for the next instruction to start are fully proven, just for some reason the PC computed by the jumps is not right now).
For RV32M, there was no goal to prove the full arithmetic - here, there are converging assertions if the outputs of the hw units are taken as expected results (so simiular to the jumps, RV32M instruction have no unexpected side effects, they "just" update their target register with a result and increment the instruction retired counter and PC trivially). In addition, we got proofs with partial data path with some bits in the source operands tied to constant values and the rest left open for proof.

Silabs-ArjanB, 2021-01-06:
Hi @tzwaenn Can you explain what the possible implications are for only having a 'bounded proof'? Also, did you have full proof for jumps on an earlier git hash? If so, what update triggered the absence of the full proof?

tzwaenn, 2021-01-07:
A bounded proof means that it is proven that no violation can happen in the given amount of n cycles from reset. This means if the shortest way to provide a failure is n+1 or nore cycles, it would not be caught. With 30 and 38 cycles from reset, one can then judge how "good" the result is (what cannpt happen in that amount of cycles from reset?). Please also not that in a later run, we got to 50 cycles from reset for the RVC jumps.
We had proofs in an eariler version that was only checking the PC update if it happend "close" to the instruction being actually executed (with the PC update in happening in decode, it can actually happen quite some time before the instruction gets executed). Activating the PC check for all cycles after a review apparently caused the convergence problems. Note that we have a tentative fix for the convergence issue running right now - hopefully leading to log with convergence for all assertions tomorrow.

tzwaenn, 2021-01-08:
Indeed, the jumps (be it RV32I or RVC) are back to converging fast with the full PC update, so we have no more inconclusive results. The result will be uploaded once the clean regression run is through.

tzwaenn, 2021-01-08:
Again, sorry for the delay - I guess it is Murphy's Law that when you need sign off results, there is a hickup in the flow and general holidays...

MikeOpenHWGroup, 2021-01-08:
> Again, sorry for the delay - I guess it is Murphy's Law that when you need sign off results, there is a hickup in the flow and general holidays...

Appreciate your perseverance Sven.  The last metre is always the longest.

tzwaenn, 2021-01-11:
With the latest report committed without bounded results, we can hopefully close this one...

MikeOpenHWGroup, 2021-01-11:
Hi @tzwaenn, do you mean the latest [OneSpin-CV32E-Results-11012021.pdf](https://github.com/openhwgroup/core-v-docs/blob/master/program/milestones/CV32E40P/RTL_Freeze_v1.0.0/Reports/Formal/OneSpin-CV32E-Results-11012021.pdf)?  This report has "partial proofs" for all RV32M properties.

Its my understanding that "partial proofs" _are_ bounded proofs.  These are acceptable if the proof depth is reported and reviewed.

tzwaenn, 2021-01-12:
Hi Mike,

for heavy arithmetic, the proof depth is usually very untelling. This is because there is a huge complexity in the pure datapath, so much so that the first instruction that can have arbitrary operand values will be the boundary you will not achieve (so the first instruction after reset in case the register file is initialized will not be reached as proof depth - add just a small added overhead in depth if an intial 0 in the registers has to be filled by some other value). Therefore, we are not reporting a bounded proof depth for M - the depth would not tell you anything.
Partial proofs means unbounded proofs are achieved for a partial proof goal. For M, this is one the one hand a completely unrestricted and unbounded proof that "just" does not check the value of X[rd], but performs all other checks required for instructions. In addition, there is then a restricted check (the source operands are restricted) which give an unbounded proof of X[rd] for that restricted set of operands. So this is something that is explicitly set up - one picks a number of bits to be free in the operands while the others have a fixed value and then runs a regression with it. We had our regression run with 4 most significant bits and 4 least siginificant bits of either operands free and the remaining bits all having the same value - so all 0 or all 1. We did not investigate which upper bound of free bits would still achieve a full proof as the data path of multiplicatoin or division was low on your list of priorities for the formal side.

MikeOpenHWGroup, 2021-01-12:
Alright, I _think_ I follow.   So, are you saying we can close this item now?

tzwaenn, 2021-01-12:
An update of the verificatoin report is pending where we added the information on the operand restriction for the M extension to clarify the partial proofs.
From my point of view, this can be closed with that update.

Silabs-ArjanB, 2021-01-18:
Hi @MikeOpenHWGroup Can this issue be closed now?

MikeOpenHWGroup, 2021-01-18:
Yes, I believe so.  As the Committer who owns the RTL Freeze decision, I believe you should have the decision on that.  My recommendation is that this issue can be closed.

Silabs-ArjanB, 2021-01-18:
@MikeOpenHWGroup , @tzwaenn core-v-docs/program/milestones/CV32E40P/RTL_Freeze_v1.0.0/Verification_openissues.md is not complete. Please fill out the sign-off dates and if there are actual exceptions/waivers/comments please list them clearly (if not remove these 'Captured as cv32e40p GitHub issues' comments

MikeOpenHWGroup, 2021-01-18:
`Verification_openissues.md` is merely a list of open issues as of the date the tag was applied and it is complete.  The sign-off those these open issues is captured either with a `WAIVED: CV32E40P` label on the issue or in the appropriate checklist.

Silabs-ArjanB, 2021-01-18:
My bad: My above comments are meant for core-v-docs/program/milestones/CV32E40P/RTL_Freeze_v1.0.0/FormalVerificationChecklist.md

MikeOpenHWGroup, 2021-01-18:
I think it _is_ complete.  Did you want a waiver for the partial proofs?

Silabs-ArjanB, 2021-01-18:
My mistake again, sorry about that (I was looking at an older commit).

closed 2021-01-18 by Silabs-ArjanB