openhwfoundation/cv32e40p #598: Make CSRW operations on HWLoop registers illegal and add cv.start and cv.end with register variant instruction

davideschiavone, 2020-12-04:
As decided during the meeting to RELOCATE the XPULP instruction and change the HWLoop ISA,

we decided to make CSRW operation illegal on HWLoop registers

Add the register variant of cv.start and cv.end instructions (@jm4rtin can you please propose an encoding?)

This TASK continues the discussion in #189

davideschiavone, 2022-12-23:
@pascalgouedo , has this done both in RTL and DOC?

pascalgouedo, 2023-01-03:
Hi @davideschiavone 
3 updates in v1.2.0 User Manual for that:
- CSRW on HWloop registers generates an illegal instruction exception,
- Added cv.start/end with register
- HWloops registers moved to custom read-only

Maybe we should moved it to Component:Doc?

davideschiavone, 2023-01-03:
is it done for the RTL? at least a simple check that you are not writing to these CSR can be done in RTL too without adding any visible cost

davideschiavone, 2023-01-03:
ok so if it is not done yet I would leave it open