openhwfoundation/cv32e40p #599: tc_clk_gating

timsaxe, 2020-12-24:
# Template for Bug Issue

Please use this template if the issue that you are reporting is a (potential) bug in CV32E40P RTL, Documentation, Verification environment or Tool and Build system.

<br>**Note**: Please file bugs related to the CV32E40P User Manual in https://github.com/openhwgroup/core-v-docs instead.
<br>**Note**: Please file bugs related to the core-v-verif verification environment in https://github.com/openhwgroup/core-v-verif instead.
<br>**Note**: Please file bugs related to the PULP tool chain in https://github.com/pulp-platform/pulp-riscv-gnu-toolchain instead.

## Bug Title
Got following message during synthesis:
Warning: Unable to resolve reference 'tc_clk_gating' in 'cv32e40p_sleep_unit_PULP_CLUSTER0'. (LINK-5)

Wondering if cv32e40p_clock_gate should be used instead?

### Component
Indicate the component in which you found a problem (and we will add the appropriate label):

**Component:RTL**

### Steps to Reproduce

MikeOpenHWGroup, 2020-12-28:
Hi @timsaxe, I'm just the verification weenie, so not the right person to answer questions about synthesis and/or physical design.  However it _looks_ to me that the [synthesis guidelines](https://github.com/openhwgroup/core-v-verif/blob/master/cv32/sim/uvmt_cv32/README.md) section of the User Manual answers your questions.   If it does not, then I would recommend generating an issue against the User Manual directly.

davideschiavone, 2020-12-28:
Hi @timsaxe , as @MikeOpenHWGroup said, have a look at the Getting Started section of the manual.

https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/getting_started.html

You need to replace the clock gating file with a file that contains your specific technology clock gating cell. The one openHW provides is just for simulation as it's tech indipendent.

Reach me on mattermost if you need more details.

Best
Davide

Silabs-ArjanB, 2021-01-04:
Hi @timsaxe I don't believe there is such module or instance called ''tc_clk_gating'' being referred to from anywhere in the RTL, so I am not sure what you are referring to. Maybe you are using a very old version of the github repos? On the latest RTL above documentation link should however answer clock gate related questions. Can you close this issue if your question has been answered or otherwise indicate which exact RTL you are using and show your file containing the clock gate?

timsaxe, 2021-01-14:
You are correct.  Was accidentally using a down-rev version

closed 2021-01-14 by timsaxe