openhwfoundation/cv32e40p #600: Input address variable "raddr_a_i" in register file is 5 bit but condition is implemented on raddr_a_i[5] index?

zeeshanrafique23, 2021-01-02:
Input address variable "raddr_a_i" in register file is 5 bit but condition is implemented on raddr_a_i[5] index, if the variable is 5 bit (4:0), how can we access 5th location of the array? 
```
  generate
    if (FPU == 1 && PULP_ZFINX == 0) begin : gen_mem_fp_read
       assign rdata_a_o = raddr_a_i[5] ? mem_fp[raddr_a_i[4:0]] : mem[raddr_a_i[4:0]];
       assign rdata_b_o = raddr_b_i[5] ? mem_fp[raddr_b_i[4:0]] : mem[raddr_b_i[4:0]];
       assign rdata_c_o = raddr_c_i[5] ? mem_fp[raddr_c_i[4:0]] : mem[raddr_c_i[4:0]];
    end else begin : gen_mem_read
       assign rdata_a_o = mem[raddr_a_i[4:0]];
       assign rdata_b_o = mem[raddr_b_i[4:0]];
       assign rdata_c_o = mem[raddr_c_i[4:0]];
    end
  endgenerate
````

As we can see in the if statement part.

Silabs-ArjanB, 2021-01-04:
Hi @zeeshanrafique23 , the input address variable "raddr_a_i" in register file is actually **6** bits, see https://github.com/openhwgroup/cv32e40p/blob/120ac3ee79ef56a57fe07dd8701cd4ee94458fd5/rtl/cv32e40p_id_stage.sv#L905

You can see here https://github.com/openhwgroup/cv32e40p/blob/120ac3ee79ef56a57fe07dd8701cd4ee94458fd5/rtl/cv32e40p_id_stage.sv#L522 how that MSB can get set.

zeeshanrafique23, 2021-01-04:
Oh got it thanks

closed 2021-01-04 by zeeshanrafique23