openhwfoundation/cv32e40p #605: Missing type declaration prefetch_buffer.sv

timsaxe, 2021-01-20:
Synopsys dc_compiler reports:

Compiling source file ../../ips/cv32e40p/./rtl/cv32e40p_prefetch_buffer.sv
Warning:  ../../ips/cv32e40p/./rtl/cv32e40p_prefetch_buffer.sv:120: the undeclared symbol 'fifo_empty' assumed to have the default net type, which is 'wire'. (VER-936)
Warning:  ../../ips/cv32e40p/./rtl/cv32e40p_prefetch_buffer.sv:140: the undeclared symbol 'fifo_full' assumed to have the default net type, which is 'wire'. (VER-936)
Presto compilation completed successfully.

I don;t know if this conforms to the style guide,  but explicit type declaration is preferable.

MikeOpenHWGroup, 2021-01-20:
Ouch!   Thanks for this @timsaxe.   Clearly needs to be fixed.

@Silabs-ArjanB, how did this get past RTL Freeze?

Silabs-ArjanB, 2021-01-20:
There is no style guide that is being followed for CV32E40P (that would be a future improvement as well). Of course we will fix this even though it is valid (and more importantly correct) Verilog as we agree that there should be no implicit wires. There are many more Lint warnings. We only made it error free, not warning free. We studied the warnings and due to the current structure of the RTL (with its unused local parameters, e.g. for user mode and PMP) we decided only to fix the ones we deemed essential for now (and do a major clean up of the RTL in a future project).

@davideschiavone Even though just adding the two wire declarations will clearly lead to a LEC equivalent design (for the frozen parameter set), before we do that we should have a flow in place to actually check this. We will need that for every commit that touches the RTL going forward. I do not have such a script that I can commit. Could you take the lead in getting that set up first (or ask somebody else to do so)?

davideschiavone, 2021-01-22:
Hello @timsaxe , thanks for reporting it!

Yes @Silabs-ArjanB, actually @MikeOpenHWGroup and I discussed it to include the LEC in the CI . I can take the lead on that.

Thanks

treforsouthwell, 2021-03-22:
It would be good to fix, our syntheiss flow flags this as an error (although it doesn't stop us getting results)

davideschiavone, 2021-04-07:
Hello @treforsouthwell , I issues a PR #633 that should guarantee that any RTL change stays Logically Equivalent to the frozen version - would you like to have a look at it? I have zero experience with LEC and Conformal

commit cc35a755 (2021-04-19): Merge pull request #634 from davideschiavone/fix605

fix #605
commit da7e83d0 (2021-04-19): fix #605
closed 2021-04-19 by davideschiavone