openhwfoundation/cv32e40p #606: Clock Gating Issue for Sleep unit

a7med7asan15, 2021-01-20:
I tried to synthesize the design without `cv32e40p_sim_clock_gate.sv` but it gave me errors 
I searched and found that I should not use it while synthesis
How could I use the technology file instead of this module? 
[image]

Silabs-ArjanB, 2021-01-21:
Hi @a7med7asan15 You should make a verilog module named cv32e40p_clock_gate that instantiates an actual clock gate cell. That new module could for example be in a file called cv32e40p_clock_gate.sv. Then you add that file to the file list used by your synthesis tool. Which cell you need depends on your technology (so we can't tell you).

a7med7asan15, 2021-01-21:
Your answer helps a lot. But Excuse me for my ignorance, I am still new to Synthesis. 
I am using the open source skywater PDK, so it has folder: CLKGATE_X1 for example
How can I instantiate this cell inside my RTL Code?

davideschiavone, 2021-02-09:
hello @a7med7asan15 , if this issue is solved, can we close it?

closed 2021-02-09 by a7med7asan15