openhwfoundation/cv32e40p #608: How to change  Pulp clusters' core to cv32e40p core

Suxiaojie-iot, 2021-01-22:
Recently, when I transplanted cv32e40p on the cluster side of Pulp, I have some difficulties——I failed to change the core, but I could not  solved it. In addition to the interface mismatch, some internal logic is also different, including the fetching and decoding stages.
In this regard, I would like to ask how to implement Pulp's cluster kernel to cv32e40p. I have read cv32e40p user  manual，but I still not  success for it.
Thank you.

davideschiavone, 2021-01-22:
Hello,

can you please describe the problem you are facing?

Suxiaojie-iot, 2021-01-22:
> Hello,
> 
> can you please describe the problem you are facing?

thanks for attention,.When I change the core of Pulp to cv32e40p,only in FC could work and I counld get "Hello world", but in CLUSTER,I changed the core to cv32e40p and simulation, it didnot work. 
In the latest version of the cv32e40p kernel, its interface is different from that in the ri5cy kernel and cannot be directly transplanted. I want to know how to configure cv32e40p to make it work properly on the cluster side of pulp.Does cv32e40p need other configuration files？

davideschiavone, 2021-01-22:
You need to re-do in the cluster side what you changed in the FC. This discussion specific to the PULP integration should be done in the PULP repository as this is the CORE repository.
From the core perspective, all the interfaces have been described in the User Manual, which are not very different from the former RI5CY. The memory protocol slighly changed and described too. The PULP team is working to adapt with bridges and/or changing their memory protocol too.

Suxiaojie-iot, 2021-01-22:
> You need to re-do in the cluster side what you changed in the FC. This discussion specific to the PULP integration should be done in the PULP repository as this is the CORE repository.
> From the core [...]

Thanks again. I haved re_do in the cluster side.It may be that I did not configure it correctly, but I did not find it. Can you tell me more specifically, or do you have a good configuration example for me to refer to? The instructions on the cluster side can only run a little bit, and then it is wrong. cv32e40p can be run on the cluster side. It should be modified a bit, but it is not clear whether it is because of the memory protocol.

davideschiavone, 2021-01-22:
I don't understand what a configuration example is.
If the problem you have is that the core is not compatible with PULP, you need to move this discussion to PULP.
They support only the old RI5CY now, and they are working to support CV32E40P

davideschiavone, 2021-01-25:
Hello @Suxiaojie-iot, if you agree that this issue is resolved (as it is off-topic for example or the issue relies somewherelse than the core), can you please close this issue?

closed 2021-01-26 by Suxiaojie-iot