openhwfoundation/cv32e40p #609: Synopsys error

timsaxe, 2021-01-25:
# Template for Bug Issue

Please use this template if the issue that you are reporting is a (potential) bug in CV32E40P RTL, Documentation, Verification environment or Tool and Build system.

<br>**Note**: Please file bugs related to the CV32E40P User Manual in https://github.com/openhwgroup/core-v-docs instead.
<br>**Note**: Please file bugs related to the core-v-verif verification environment in https://github.com/openhwgroup/core-v-verif instead.
<br>**Note**: Please file bugs related to the PULP tool chain in https://github.com/pulp-platform/pulp-riscv-gnu-toolchain instead.

## Bug Title
When synthesizing cv32e40p Synopsys reports the following:

Compiling source file ../../ips/cv32e40p/./rtl/cv32e40p_pmp.sv
Error:  ../../ips/cv32e40p/./rtl/cv32e40p_pmp.sv:671: redeclaration of symbol IDLE as constant. (VER-513)
Error:  ../../ips/cv32e40p/./rtl/cv32e40p_pmp.sv:671: symbol IDLE must be a constant or parameter. (VER-260)
*** Presto compilation terminated with 2 errors. ***

### Component
Indicate the component in which you found a problem (and we will add the appropriate label):

**Component:RTL**: For issues in the RTL (e.g. for files in the rtl directory)

### Steps to Reproduce
Please provide:
Using tag _v1.0.0

Search in the relevant files reports:
Search "IDLE" (6 hits in 2 files of 3 searched)
  C:\Users\saxe\Downloads\cv32e40p_pkg.sv (2 hits)
	Line 206: typedef enum logic {IDLE, BRANCH_WAIT} prefetch_state_e;
	Line 208: typedef enum logic [2:0] {IDLE_MULT, STEP0, STEP1, STEP2, FINISH} mult_state_e;
  C:\Users\saxe\Downloads\cv32e40p_pmp.sv (4 hits)
	Line 671:    enum logic {IDLE, GIVE_ERROR} data_err_state_q, data_err_state_n;
	Line 679:          IDLE:
	Line 689:                data_err_state_n = IDLE;
	Line 697:           data_err_state_q <= IDLE;

If I change 'enum logic {IDLE, GIVE_ERROR} data_err_state_q, data_err_state_n;' to
'enum logic {IDLE_ERROR, GIVE_ERROR} data_err_state_q, data_err_state_n;'
and update the file accordingly the error message goes away, suggesting that synopsys does not accept the same token in two different enums (I notice that in the pkg the first enum has IDLE and the second IDLE_MULT suggesting someone ran into this)

It seems unreasonable to require different tokens even though the enums are different, so I suspect this is Synopsys issue, not a language issue.

Silabs-ArjanB, 2021-01-25:
Hi @timsaxe thank you for your issue report. The CV32E40P does not contain a PMP. That file should not be synthesized as described here https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/intro.html#synthesis-guidelines

davideschiavone, 2021-01-25:
Hi @timsaxe ,  I guess you are synthesizing the core using the src_file instead of the manifest file. And the same for simulation too. I suggest you open a new issue (and assign it to me or make directly a pull request) to aling the src_file and Bender file to the manifest.

timsaxe, 2021-01-26:
Hi Arjan,

Davide also responded to this.  The issue is triggered by having the PMP file in the src_files.yml manifest, which means that the generate-scripts includes it.  I will create a PR removing PMP from src_files.yml.

Tim

davideschiavone, 2021-02-09:
Hi @timsaxe , can we close now this issue?

timsaxe, 2021-02-09:
We can.  I think I did, but if that didn’t work you can close it.

closed 2021-02-09 by timsaxe