openhwfoundation/cv32e40p #612: some question in rtl,may be it is a bug

yangguoming, 2021-01-29:
In source rtl code <cv32e40p_prefetch_controller.sv>
```
  assign fifo_valid  = !fifo_empty_i;
  assign fifo_push_o = resp_valid_i && (fifo_valid || !fetch_ready_i) && !(branch_i || (flush_cnt_q > 0));
```
what is the mean of " if fifo is not empty,then we can push fifo",it is not obey the logic, I think it is should be "if fifo is not full,then we can push the fifo",so the code should be the following code?
```
 assign fifo_has_space   = !fifo_full_i;
 assign fifo_push_o = resp_valid_i && fifo_has_space && !fetch_ready_i && !(branch_i || (flush_cnt_q > 0));
```
Others,i can not find the fifo full signal which should connect the controller.Thanks!

MikeOpenHWGroup, 2024-06-04:
Hi @davideschiavone and @pascalgouedo.  I know it has been  years, but I think that at least one of you should be able to answer @yangguoming's question and close this issue.

pascalgouedo, 2024-06-04:
cv32e40p_prefetch_controller doesn't show any incorrect behavior after a lot of non-regressions (and even a lot of test chips).
It is fully covered in simulation without any waiver, this is either a performance problem not visible in simulation or not a valid assumption at all.
I will let @davide analyze it in details and answer as he is maybe the one who written this assign 😁.

davideschiavone, 2024-06-04:
agree with @pascalgouedo , we can close this issue - @yangguoming let us know if you find any bug and feel free to reopen the issue

closed 2024-06-04 by davideschiavone