openhwfoundation/cv32e40p #618: Make Hardware Loop end-instructions point to the instruction after the last loop instruction

flip1995, 2021-02-08:
### Background information

The current definition of CORE-V hardware loop instructions introduce several challenges to both tool writers and users, and we feel before people are relying on the current definition, we can make a small change which can help greatly.

Currently the instructions `cv.setup`, `cv.setupi` and `cv.endi` get the location of the last instruction in the HWLP:

```s
entry:
    cv.endi 0, end
    ...
end:
    addi t0, t0, 1
    ...
```

This presents two challenges:

1. To compilers, we have to introduce unnecessary "control flow" into the instruction stream. Typically you can only refer to the start of a basic block (e.g. a branch target), with the compiler emitting labels into assembly at these points. With the current definition we have to break up the instruction stream into two blocks, with one containing the final instruction, and maintain block placement such that these always appear adjacent with no other instructions placed between. Additionally since we have split up the block, future compiler optimization may be hindered.

2. To end users it introduces the possibility of either misreading how long the loop is, or if they are writing assembly, they may mis-label the loop endpoint, and end up with one fewer or greater instruction as needed. The intuitive reading would be that the loop ends before the loop, not one instruction afterwards.

### Solution

To solve both of these we propose that the `end` label should point to the instruction **after** the last instruction in the loop. This would simplify writing optimizations with HWLPs, without tripping over hard to notice errors, that can come from this.

---

cc @jeremybennett @jessicamills @davideschiavone as discussed in the meeting last Friday (2020-12-04).

I'm happy to change the documentation for this, but I think this has to be implemented in the ISA first?

davideschiavone, 2022-12-23:
@pascalgouedo - is this solved both in RTL and Doc?

pascalgouedo, 2023-01-03:
Hi @davideschiavone 
Yes included in v1.2.0 User Manual.

closed 2023-01-03 by davideschiavone