openhwfoundation/cv32e40p #619: Add support for stoptimer and stop counter

gmartin102, 2021-02-09:
The Risc-v External Debug Support Document refers to to bits in the DCSR register for stopping timers and counters during debugging.  The CV32e40P has these bits, but they are not available at the top level.  The signal debug_halted_o is available and can be used to stop timers and counters, but the bits in the dcsr give greater control.  The MTIME/MTIMECMP interrupt servicing is one area that would benefit from the ability to stop the timer while halted for debug.

davideschiavone, 2021-02-09:
Hello @gmartin102 - it sounds as an "OK" modification to me as although it changes the core interface, it just exposes more signals, thus from a verification and logical equivalence point of view it should be ok.
I leave the Core Task Group Chair taking a decision on this @Silabs-ArjanB

Silabs-ArjanB, 2021-02-09:
Hello @gmartin102 I assume that you are talking about adding support for the _stoptime_ and _stopcount_ bits, which are currently implemented as WARL bits tied to 0 (as allowed in the RISC-V Debug specification). As your proposal is not a bug fix nor completion of the PULP ISA this would be a not-LEC-equivalent addition of a new feature, which we agreed not to allow after RTL freeze (see https://cv32e40p.readthedocs.io/en/latest/core_versions/).

gmartin102, 2021-02-10:
I understand this position, the functionality I wish to support can still be provided with the available signals on the cv32e40p  albeit with reduced ability to control the function as intended by the dcsr specification.

davideschiavone, 2021-02-25:
Hello @gmartin102 , if the issue is solved, can we close it?

davideschiavone, 2021-02-25:
if you think this feature is valuable, open an issue in our next 3 cores as: 

CVA6 https://github.com/openhwgroup/cva6

CV32E40S https://github.com/openhwgroup/cv32e40s

CV32E40X https://github.com/openhwgroup/cv32e40x

closed 2021-02-25 by davideschiavone