openhwfoundation/cv32e40p #626: Variable `ram_data_gnt` is driven by procedural and continuous assignment

dawidzim, 2021-03-01:
Hey!
Variable `ram_data_gnt` is driven by procedural and continuous assignment in file:
https://github.com/openhwgroup/cv32e40p/blob/master/example_tb/core/mm_ram.sv
line 517 and 605,617

MikeOpenHWGroup, 2021-03-01:
Hi @dawidzim, thanks for the issue.  Just FYI, the `example_tb` in cv32e40p is not considered a "production" testbench.  Its purpose is to support designer-testing without the overhead that comes from a regressionable verification environment such as [core-v-verif](https://github.com/openhwgroup/core-v-verif).   I have assigned this issue to @davideschiavone.

davideschiavone, 2021-03-01:
Hi @dawidzim ,

thanks for reporting this issue.

I made a PR here: #628

let me know if this solves your issue

dawidzim, 2021-03-02:
Hey @davideschiavone 
Thanks, I think this PR will solve this problem. After merge I will close this issue

davideschiavone, 2021-03-02:
Thanks! Merged

PR #628 (2021-03-02): :bug: fix #626
This should fix #626 and add the randomness in the grant bus as suggested by @Silabs-ArjanB in #625 .

I copied from https://github.com/openhwgroup/core-v-verif/pull/486/commits/ca1181d2b4f8ea36dfaf457d941db6ffdc1305eb
commit 4c293526 (2021-03-02): :bug: fix #626
commit 83ef8aa6 (2021-03-02): Merge pull request #628 from davideschiavone/fix_delay_clk

:bug: fix #626
closed 2021-03-02 by davideschiavone